pybricks/support #2087: [Bug]

Chrissy6576, 2025-02-22:
**Describe the bug**
What is the problem?

I had code working Feb 1st on a Spike Prime hub, which worked on two different computers and two different hubs.  It had a fully functioning menu structure I had used from the Monongahela Cryptid cooperative code that allowed me to select code orders on our Spike prime hub.  This worked Feb 1, and we just brought up the same code for the first time since our competition.  We even re-formatted two different hubs. 

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

I pair the spike prime via Bluetooth and press play to load the program.  Now I just get a small square on the hub instead of my menu.  Did the code base change?

**Expected behavior**
What did you expect to happen instead?

I pair the spike prime via Bluetooth and press play to load the program.  Now I just get a small square on the hub instead of my menu.  Did the code base change?

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

[image]

dlech, 2025-02-23:
Are you using `code.pybricks.com` or `beta.pybricks.com`?

Did you update the firmware on your hub since it was working?

laurensvalk, 2025-02-23:
If the program doesn't start, it usually explains what is happening in the bottom output pane. (You might have to drag it upwards to reveal it if you hid it earlier.) 

What error messages do you see?

If you're loading this for a new computer/hub, is it perhaps possible that the motors and sensors are not yet plugged in, or could one of the submodules still be missing?

BertLindeman, 2025-02-23:
Did you also load the file `supportcode.py` into pybricks in chrome?