pybricks/support #2096: [Bug] drivebase - curve by an given angle and radius

Sirus-7, 2025-02-28:
**Describe the bug**

Technic Hub - Pybricks Beta v3.6.0b5

If in drivebase - curve by an given angle and radius, _continue_ is selected instead of _hold_.

**Screenshots**
[image]

laurensvalk, 2025-02-28:
Thank you very much for reporting! We'll fix this as soon as possible.

Kelv68, 2025-03-01:
Also Veer does not work with continue, brake, and coast.

laurensvalk, 2025-03-01:
Indeed. These will be fixed too.

The issue is that the generated code is similar to:

```
robot.arc(500, distance=250, Stop.COAST)
```

But there can't be a positional argument after another keyword argument.

We will fix it by generating this code instead:

```
robot.arc(500, distance=250, then=Stop.COAST)
```

This will automatically come to an updated beta version soon.

Sirus-7, 2025-03-02:
I think it would be a better solution if we always use "function(parameter name=value, ...)"
like this: "robot.arc(arc=500, distance=250, then=Stop.COAST)."

laurensvalk, 2025-03-02:
It would be nice to have an option in the settings so you can choose. 

Turning it on would make it easier to read, but the program also becomes quite a bit bigger. So then you can choose.

Perhaps in a future release 🙂

Sirus-7, 2025-03-02:
[image]

NameError: name 'Stop' isn't defind

BertLindeman, 2025-03-02:
> [image]
> 
> NameError: name 'Stop' isn't defind

Hi,

I just updated beta.pybricks.com and installed the new firmware: ```v3.6.0 (Pybricks Beta v2.6.0-beta.6)```
Did not see this error on: 

[image]

laurensvalk, 2025-03-02:
> [image]
> 
> NameError: name 'Stop' isn't defind
> 
> 

Does this occur with a block program? If so, can you share it?

Sirus-7, 2025-03-02:
No, this occur with the text base program.  v3.6.0 (Pybricks Beta v2.6.0-beta.6)

[image]

BertLindeman, 2025-03-02:
Did you forget to define Stop?
```python
from pybricks.parameters import Stop
```

Bert

Sirus-7, 2025-03-02:
You're right Bert. But why isn't this visible in the text in block programming?

[image]

closed 2025-03-02 by Sirus-7