pybricks/support #2117: [Bug] Hub goes to a previous code

cayiranil, 2025-03-13:
So, I was using my Technic Hub, and I uploaded a code into it(code 1).  A while later, I uploaded another code (code 2)

cayiranil, 2025-03-13:
Accidentaly submitted it without properly finishing (new to github and accidentaly marked triage somehow) it:
After toying around with it for some time, the hub went back to code 1. I don't know how it happened, it seemed to happen out of nowhere. I don't know how to reproduce this problem.

laurensvalk, 2025-03-13:
Does this still happen if you download your program to the hub again?

Programs are saved when the hub shuts down normally. Is it possible that you were coding and pulled the batteries before shutting down normally? Then the newest program won't be saved and it will still have the last properly saved program on it.

cayiranil, 2025-03-13:
I couldn't download the program again at the time, and I didn't remove the batteries from the hub. And I did shut the hub down normally.

laurensvalk, 2025-03-14:
Could you please share the program that would not be saved, and if you can, the other program as well?

I'd like to try if I can reproduce it and possibly fix it.

cayiranil, 2025-03-14:
The program was saved normally, but a while later it went to the previous code. (will code files in txt because github didn't accept .py)

[code1.txt](https://github.com/user-attachments/files/19246299/code1.txt)

[code2.txt](https://github.com/user-attachments/files/19246303/code2.txt)

laurensvalk, 2025-03-14:
Thanks! Which firmware version were you using? Did you recently update it or have you been using it for some time? Here's how you find out:

[image]

cayiranil, 2025-03-14:
I was using v3.5.0 but I recently updated the firmware to the beta version hoping that it would get solved.

cayiranil, 2025-03-16:
I was using v3.5.0 but after this incident I updated it to the beta version just to be safe.