pybricks/support #2123: [Question] How to clear hub.ble.observe received data?

JorgePe, 2025-03-18:
I'm using linux hcitool to broadcast messages in the Pybricks broadcast format.
https://github.com/JorgePe/EV3-Pybricks-Messaging

I've build a Technic Hub StarterBot and adapted the example code to react to 3 string messages ('MOV', 'ROT', 'BAK'):

```
hub = TechnicHub(broadcast_channel=0, observe_channels=[1])

sensor = ColorDistanceSensor(Port.C)
left = Motor(Port.A, Direction.COUNTERCLOCKWISE)
right = Motor(Port.B, Direction.CLOCKWISE)
robot = DriveBase(left, right, 56, 128)

while True:
    rcv = 0
    rcv = hub.ble.observe(1)
    if rcv is None:
        hub.light.on(Color.RED)
    else:
        hub.light.on(Color.GREEN)
        if rcv == 'MOV':
            robot.straight(250)
        elif rcv == 'ROT':
            robot.turn(90)
        elif rcv == 'BAK':
            robot.straight(-250)

    # Broadcasts are only sent every 100 milliseconds, so there is
    # no reason to call the observe() method more often than that.        
    wait(100)
```

I'm using MINDSTORMS EV3 to advertise those messages.

I start advertising and immediately send an HCI command to stop advertising but I am not sure exactly how much time it takes until it really stops. But I was assuming less than 100 ms so the Hub would only receive one message.

But several times (not always) 'ROT' action takes place twice.
So I added a 500 ms pause and it works fine. The other two actions take longer to execute so no problem there.

I am also aware of #1704 but don't think that's my "problem" because when I run this script on another Technic Hub:

```
hub = TechnicHub(broadcast_channel=1)
while True:
    hub.ble.broadcast("ROT")
    wait(50)
    hub.ble.broadcast(None)
    wait(5000)
```

since I'm disabling broadcasting after 50 ms the observer should only receive 1 advertisement, right?

But at least one-in-five my StarterBot rotates twice. Since it doesn't always happen I suppose it's not a question of receiving more than one advertisement (but with BLE I'm never sure). Some buffer/cache perhaps?

Am I doing something wrong in the StarterBot script?
Or am I making wrong assumptions about the way broadcast works?

BertLindeman, 2025-03-18:
Hi Jorge,

Did you consider to replace the "ROT" command by either "UP" or "DOWN" or something like that?
That would make it irrelevant it the message was received more than once?

Nice setup though.

Bert

JorgePe, 2025-03-18:
Thanks for answering me Bert.

In my EV3 I did try to replace the stopping of the advertisement on channel 1 with a broadcast on channel 0, it didn't worked. But never tried what you suggested, my fault (maybe because I didn't want my scripts to keep sending meaningless data)

So... instead of stopping the advertisement  I am now sending a False on channel 1. And it works as expected, no need for a delay on the StarterBot side.

In the other Technic Hub I now have this loop:

```
while True:
    hub.ble.broadcast("ROT")
    wait(50)
    #  hub.ble.broadcast(None)
    hub.ble.broadcast(False)
    wait(5000)
```

and it also works as expected.
So I cannot assume that

```
hub.ble.broadcast(None) 
```

takes place immediately. Or maybe it takes but the receiver sometimes needs some kind of notification that a new value arrived clear the old one? I can live with that.

Thanks for your time. Great support!!

JorgePe, 2025-03-18:
Working great :D

[image]

Closing this.

closed 2025-03-18 by JorgePe