pybricks/support #2128: ``` Connection Error: Connection attempt failed. ``` Cant connect my new hub to pybricks

ollyoYT, 2025-03-20:
Whenever I try to download the Pybricks firmware the error message ```
Connection Error: Connection attempt failed.
``` always comes up. I have tried it many times and am quite disappointed as the technic hub was costly. Please help me if you can :)
Thank you!

dlech, 2025-03-20:
Can you please give more information on how exactly to reproduce this problem? 

* What operating system?
* Are you using https://code.pybricks.com ?
* What are the exact steps that lead to this? For example, I click on... then I clicked on... then pressed and held the button on my hub...

ollyoYT, 2025-03-21:
@dlech Hi there! Thanks for replying to my problem. Sorry i didn't specify further. Here are the answers :) 
I am on windows 10
I tried both code.pybricks.com and beta.pybricks.com, neither work
The exact steps are as follows: Step one, I click onto "install Pybricks firmware". Step two, I selected the hub, accepted the licenses, named the device and watched the video. Step three, I held down the green button on the hub, it turned purple so i clicked install (still holding the button) Step four, I noted that the light on the hub flashed multiple colours. Step five, when the hub stopped flashing the error message "Connection Error: Connection attempt failed" popped up on my screen. Step six, I tried multiple times and still nothing so I can to github to report my issue!
Thanks again, I really hope you can help me. :)

dlech, 2025-03-21:
Once the hub starts flashing multiple colors, you can let go of the button. Some of the hubs don't like it if you hold the button too long. Low batteries can also cause problems when flashing firmware. And on the City Hub, having anything plugged into an I/O port will cause flashing firmware to fail.

dlech, 2025-03-21:
Yes, could be a Bluetooth adapter issue. Intel Wireless + Bluetooth is notorious for not working well with Bluetooth low energy.

Some people have better luck flashing the firmware, e.g. with their Android phone, then they can still use Pybricks Code on their computer.

ollyoYT, 2025-03-21:
@dlech Thanks a million for all of your help. I'll defo put it into action. Bye

closed 2025-03-21 by ollyoYT