pybricks/support #2145: [Bug] pybricks 3.6.1 driveBase.settings: "invalid argument"

bcfletcher, 2025-04-03:
we are getting "Invalid Argument" for 

    driveBase.settings(turn_acceleration=turnAccel,
                       turn_rate=turnRate)

when turnRate >= 612

this was not an issue with 3.5. we used 1000 without problems

we have tried this on two different Spike Prime hubs with identical results

turnAccel doesn't see to have this a-problem

dlech, 2025-04-03:
`turn_rate` has units of degrees per second. So 612 degrees per second would mean that the robot would have to completely turn around 1.7 times per second. This is very fast. So the error is probably saying that the motors can't make the robot turn faster than that, even at top speed.

`turn_acceleration` has units of degrees per second per second. The motors can accelerate to the max speed in less that one second, so it makes sense that larger numbers would be allowed here without exceeding the physical limits of the robot.

bcfletcher, 2025-04-03:
ok but 950 was the value we using 3.5 without issue. in 3.6.1 it causes "invalid argument". 

so even if 950 or even 612 is too much the error shouldn't be "invalid argument".  it should be "valid range exceeded" or something like that

laurensvalk, 2025-04-03:
> ok but 950 was the value we using 3.5 without issue. in 3.6.1 it causes "invalid argument".

These settings haven't changed in a long time (since version 3.3). What are your drive base size and wheel diameter?

A wide robot with tiny wheels cannot turn very fast (and vice versa), so these robots will hit the limit sooner. Did you use the same design as before, or perhaps it is a new design?

bcfletcher, 2025-04-03:
yes the robot designed changed to wider box style but with 62.4 wheels vs. the smaller spike blue ones

so i still don’t understand your rationale. if 950 worked on previous robot but the new robot immediately rejects 950 before it has started to even move doesn’t make based on your comment above.

the error occurred trying to set  the value not from the actual turn command

so the robot at this point would have no idea of how wide it was or the size of the wheels

bcfletcher, 2025-04-03:
oh wait, so it know they based solely on the wheel diameter and drive base setting during hub initialization?

so it does a calculation a d says i can’t do that?

laurensvalk, 2025-04-03:
It works out how fast the robot can turn given that the motors can go at most 1000 deg/s. A wider drivebase or one with smaller wheels will have a lower top speed. 

This is ultimately intended to help you - it's better to be warned up front that an ambitiously high speed isn't going to work, rather than finding out while running or relying on an incorrect value in. 

I do agree that we could make the messaging more helpful. I've opened a dedicated issue to improve the error message.

bcfletcher, 2025-04-03:
thanks for the explanation

is there a way to ask the robot what the max calculated values are ?

does the calculation apply only turn_rate or are there also calculations for turn_accel, straight_speed, straight_accel max values?

BertLindeman, 2025-04-03:
From the doc: ```The initial values are automatically configured based on your wheel diameter and axle track. They are selected such that your robot drives at about 40% of its maximum speed.```
Could you calculate from there?

laurettamorris, 2025-04-04:
We are also having an issue with  the curve command with the drive base.  Here are screenshots of our setup as well as the blocks implementing the curve.  You can see that they have the ! on them showing there is an issue but I am not getting a written error message.  We are implementing a very small curve. 

[image]

[image]

laurettamorris, 2025-04-04:
Thanks!  I am new I really appreciate the help.  I see what you mean on all acocunts.

bcfletcher, 2025-04-05:
Laurens,

I submitted a pull request (I hope) for my edits  to robotics.py with improved wording for the DriveBase.settings