pybricks/support #2147: [Bug] ``` GATT operation failed for unknown reason. ```

agni2012, 2025-04-05:
**Describe the bug**
Nothing; It just shows that message, continuing running the message.

**To reproduce**
It is random; it happend just now.

**Expected behavior**
For the error message to not happen

**Screenshots**
No. I just showed a message.
Cant reproduce

dlech, 2025-04-06:
If the problem isn't reproducible, then there isn't much we can do. If you ever find a way to reproduce it, please do let us know.

closed 2025-04-06 by dlech

agni2012, 2025-04-08:
ok thanks

McCavity, 2025-05-05:
I'm having the same issue since today:

**Describe the bug**
The issue started, after I added a [88007 Color & Distance Sensor](https://www.bricklink.com/v2/catalog/catalogitem.page?S=88007-1) to my [PyBricks Starter Bot (Technic Hub)](https://pybricks.com/learn/building-a-robot/#technic). The Bot is almost identical to the one in the example, I only had to improvise on a few parts I had missing (especially the wheels are constructed slightly different) and it already has a light attached to it on port D, is connected to an X-Box Controller which can start the "run a square" program by pressing the "X" button, swith the light on and off by pressing the "A" button and while no program is running, you can control both wheels individually with the vertical axes of the two joysticks. Only thing I added today was the sensor on port C, before that, everything worked as expected.

I modified the program to accomodate the sensor but when I tried to "run" it (e.g. send it to the hub and start) nothing happened for a while - and the this same error message appeared (see als the attached screenshot9. Since then I cannot send any program (I removed the sensor again and tried to install an earlier version) to the hub anymore, every time I try to send it fails. The program that remains on the hub seems to be the working one before I tried to include the sensor, because when I start the hub without PyBricks, it'll pair happily with the X-Box Controller and execute the commands sent by it.

I also tried reloading the browser completely, I did not (yet) try to completely restart the browser - I'll do that next after this bug report.

**To reproduce**
- Switch on Technic Hub (ok)
- Select Technic Hub in PyBricks Bluetooth connection dialogue (ok)
- attempt to upload program (takes about half a minute, sometimes a circle appears in the "run" symbol signallling "downloading... xy%" with the percentage usually stopping at 15% but at one time it progressed to 79%. The it stalls for about 30 seconds, after that the error message appears and the Bluetooth Pairing is broken. i.e. the Technic Hub goes back to double-blink and in PyBricks it can be selected for pairing again.

**Expected behavior**
Transfer and start programs normally or at least give more information on what went wrong

**Screenshots**
[image]

McCavity, 2025-05-05:
Update: restarting the browser made the problem go away, I could successfully write the program again - without the sensor, for the moment. The progress circle went full this time and the hub disconnected (the well known behaviour when using an X-Box Controller). Screenshot from when the program was being transferred:

[image]

McCavity, 2025-05-05:
Update #2: as soon as I attempt to integrate the sensor, the problem reappers. Could be I have a faulty sensor? It was sold as "new, sealed" and I just unpacked it today, if I attach it to the bot it will cycle through all colours and then switch on all of them, which seems normal behaviour to me. Unfortunately I didn't have any sensor befor so I can't compare to a known-good situation.

dlech, 2025-05-05:
Hi @McCavity, This is a closed issue about Bluetooth, so posting here won't likely get much attention. Since it seems like your Bluetooth issue sorted itself out, I would suggest starting a new [discussion](https://github.com/orgs/pybricks/discussions) to see if we can figure out the sensor issue.

McCavity, 2025-05-06:
@dlech thanks for the advice! For the moment, the problem seems to have disappeared. After letting it sit for the night (I didn't even restart the browser after the error appeared for the second time yesterday) I paired the hub just a few a minutes ago and uploaded the program as-is - with sensor attached and the sensor configuration in the program. One thing I noticed was how fast the program was uploaded to the hub. Yesterday I'd estimate it took around 10 to 12 seconds for the - very short - program to upload, today it was finished after only three seconds.

I'd draw the conclusion that the sensor isn't faulty at all but that I rather might have experienced some strange kind of performance problem where the upload for some reason took too long and ran into a timeout.

Thanks anyway for pointing me to the discussion page, I'll open one if the error should come back - maybe it can be improved to give some more meaningful information.

Thanks again!