pybricks/support #2150: [Bug] Cannot run programs when Bluetooth is turned off starting with release 3.6.0b2

garyiwu, 2025-04-07:
**Describe the bug**
In pybricks firmware release [3.6.0b1](https://github.com/pybricks/pybricks-micropython/releases/tag/v3.6.0b1) a feature was added to allow Bluetooth to be toggled on and off.  This worked well.

In every release tested after that ([3.6.0b2](https://github.com/pybricks/pybricks-micropython/releases/tag/v3.6.0b2) through [3.6.1](https://github.com/pybricks/pybricks-micropython/releases/tag/v3.6.1)), if Bluetooth is toggled off, then we cannot run the loaded program.  Pressing the main button only triggers a brief flicker in the primary matrix display and nothing else.  Once we toggle Bluetooth back on then pressing the main button executes the loaded program normally.

This was reproduced on multiple Spike Prime hubs.

**To reproduce**
Steps to reproduce the behavior:
1.  Write/load a simple program, say one that just makes a beep.
2.  When Bluetooth is on, press the main button to execute the program.
3.  Toggle Bluetooth to be off.
4.  Press the main button and observe that the program does not run.

**Expected behavior**
We expected that we can run the currently loaded program irrespective of whether the Bluetooth is on or off.

BertLindeman, 2025-04-07:
Hi,

I flashed the beta firmware to a Spike Prime hub.
```('primehub', '3.6.1', 'ci-release-86-v3.6.1 on 2025-03-11')```

1. Connect to the hub with bluetooth and run this program:

```python

from pybricks.hubs import ThisHub
hub = ThisHub()
try:
    hub.speaker.volume(20)
    hub.speaker.beep()
except:
    print("no speaker")
    pass
```

 2. disconnect the hub from the pc
 3. press the bluetooth button to disable bluetooth. The BT button does not longer flash blue.
 4. press the center button on the hub and it beeps, so the program runs.
 5. press the bluetooth button to re-enable bluetooth. BT button flashes blue
 6. press the center button on the hub and it beeps, so the program runs.

Can not reproduce your scenario.

Is your program different from mine and / or do you use another version firmware?

I am curious.

Bert

garyiwu, 2025-04-07:
Thanks for trying it!  With your input, I was able to narrow the issue down; looks like the problem occurs when `broadcast_channel` is specified.

With Bluetooth off, this program fails to run:

```
from pybricks.hubs import PrimeHub
hub = PrimeHub(broadcast_channel=0)
hub.speaker.beep()
```

Whereas this program runs correctly with Bluetooth off:
```
from pybricks.hubs import PrimeHub
hub = PrimeHub(broadcast_channel=None)
hub.speaker.beep()
```

BertLindeman, 2025-04-07:
Ah I see.

Well it seems reasonable that if you ask for bluetooth connection (with broadcast_channel), the program needs bluetooth.

garyiwu, 2025-04-07:
Yeah, not sure what the expected behavior is in this case.  We had just upgraded from 3.5.0 where the default value for `broadcast_channel` used to be `0` and was explicitly defined in our code, even though we don't make use of `hub.ble.broadcast()`.  Please feel free to close the issue if this is considered expected behavior.  Thanks!

laurensvalk, 2025-04-07:
Thanks @BertLindeman for helping and @garyiwu for the additional info - it's great that you went back to try it on older versions too.

Indeed, when Bluetooth is turned off, Bluetooth functions cannot be used and the program stops. If you don't need broadcasting in your code, you can just do:

```python
hub = PrimeHub()
```

> not sure what the expected behavior is in this case

We meant to clarify it by choosing `None` by default for broadcasting since no channel is active by default (not even `0`), but I can see that the change may appear confusing in this special case.