pybricks/support #2152: [EV3 - help wanted]: Use DMA for display SPI

laurensvalk, 2025-04-08:
We currently have an [IRQ implementation](https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/drv/display/display_ev3.c) for the EV3 display. Since we have to write big continuous buffers we should be using the EDMA for this. All the necessary config [can be found here](https://github.com/pybricks/ev3rt-lib/blob/master/target/ev3_gcc/drivers/lcd/src/lcd_spi.c) but I could not get it to work in our driver.

The DMA does not seem to write any data, or perhaps reads only zeros. Could it be that the MMU/Cache is not configured properly? Do we need to do this in our firmware or maybe in u-boot? Note that that EV3RT driver is in turn based on the ev3dev driver. I have taken a minimal combination of both, stripping away almost all of the Linux stuff.

On a related note: I got some unusual behavior when I originally had the buffers as large static arrays in the driver module. They seemed to cause out of bound reads or writes, as adjacent variables (according to the elf map file) got bad values. I was able to work around this by putting the buffers in a dedicated spot with the linker file. Putting them there is probably not a bad thing, but we really shouldn't have to do that just to work around memory problems, so maybe there is something deeper going on here.

erik-thompson-seattle, 2025-04-14:
I'll start diving into this once I get my build system setup.

dlech, 2025-04-14:
> but we really shouldn't have to do that just to work around memory problems,

Actually, for DMA, yes you should have to do things like this. It is modifying the memory without the CPU having any knowledge and vice versa. In Linux there is a special CMA allocator for creating large, continuous buffers for use with DMA. Carving out a dedicated RAM hardware address range works too.

>   as adjacent variables (according to the elf map file) got bad values

This is very much a typical thing that happens when memory is in the same CPU cache line that is being accessed by DMA. A typical mode of failure is something like this: DMA gets one copy of a block of memory (aka cache line), modifies it and writes it back while at the same time the CPU modified something else in the same block. The CPU's changes get written over with what the DMA read before.

> I'll start diving into this once I get my build system setup.

Don't be shy about asking questions. I have quite a bit of experience with this hardware on Linux at the kernel driver level. 😄

laurensvalk, 2025-04-14:
The memory issues commented on above were happening with the current IRQ driver, not the DMA attempt (which did not work at all).

The DMA implementation used in EV3RT does indeed call a cache clean function, but maybe we are missing some other MMU setup steps early on to make this work properly.

The same TI library functions are available in our repository already so we can call them if we need to.

While u-boot could play a role, I did confirm that the EV3RT driver does work with the U-boot version we are using and vice versa, so it is not as simple as that. 

Thanks @erik-thompson-seattle for taking a look!

laurensvalk, 2025-05-10:
We've just posted a [standalone version of the Texas Instruments StarterWare for EV3](https://github.com/pybricks/AM1808_StarterWare_1_00_03_03). Other than the fact that this runs with u-boot, this gets very close to how TI expects things to work on this processor.

The following example works:

```c
examples/evmAM1808/uart/uartEcho.c
```

These two do not:

```c
examples/evmAM1808/uart_edma/uartEcho.c
examples/evmAM1808/cache_mmu/uartEdma_Cache.c
```

@dlech, Maybe this could help us figure out why DMA is not working, entirely independently from Pybricks driver code?

laurensvalk, 2025-05-10:
I'm able to run some code without u-boot as well, though not yet a full example. I'll try to get that running and push it above. That should get us as close to the bare TI examples as it gets.

laurensvalk, 2025-05-10:
Now updated the repo to boot the basic IRQ UART echo example without u-boot altogether, so things are very close to the original TI code now. Ideally, we could now find why `examples/evmAM1808/uart_edma/uartEcho.c` is still outputting garbage. 

(It looks like the bootloader is already setting up enough DDR to get things working, so maybe we technically don't need u-boot. It would still be nice to include so people can still use ev3dev effortlessly by just plugging in the microSD card though.
Does anyone know if the EV3 bootloader source was ever released? Based on its output and jump address, I have a hunch that it is based on the bootloader example in the aforementioned StarterWare repo. We don't really need to know this. It's just to get a slightly better understanding of how things might work and which things are already enabled.)

laurensvalk, 2025-05-10:
Though on the bright side, the included edma example for an internal transfer (`examples/evmAM1808/edma/edmaTest.c`) does work, so maybe things are pretty close.

```
Welcome to EDMA/QDMA example application.

edma3Test: Data write-read matching PASSED.
edma3Test PASSED.

EDMA/QDMA application is successfully completed.
```

> > Does anyone know if the EV3 bootloader source was ever released?
> 
> I guess you mean the first-stage bootloader that is in the EEPROM and not the second stage bootloader (U-boot)? No, that was never released. We can use `i2cdump` on ev3dev to read it though and reverse engineer it if we need to.

Right. I don't suppose we truly need it at the moment.

BertLindeman, 2025-05-10:
> We've just posted a [standalone version of the Texas Instruments StarterWare for EV3](https://github.com/pybricks/AM1808_StarterWare_1_00_03_03). Other than the fact that this runs with u-boot, this gets very close to how TI expects things to work on this processor.
> 
> The following example works:
> 
> examples/evmAM1808/uart/uartEcho.c
> These two do not:
> 
> examples/evmAM1808/uart_edma/uartEcho.c
> examples/evmAM1808/cache_mmu/uartEdma_Cache.c
> [@dlech](https://github.com/dlech), Maybe this could help us figure out why DMA is not working, entirely independently from Pybricks driver code?

Trying to make that test. Missing ` ../pybricks-micropython/tools/metadata.py` and maybe readelf.py?
Tried to fake some, but the result of `make deploy` now is ```make: *** [Makefile:175: deploy] ```
Obvious I miss some package?

BertLindeman, 2025-05-10:
Dirty trick: copied ```cp -p  /home/bert/py/pybricks/git_clone/pybricks-micropython/tools/metadata.py ../pybricks-micropython/tools/``` And the build runs ok, but the EV3 stays at starting.
Obviously not a compatible setup here.

laurensvalk, 2025-05-10:
Yes, that is expected. It is very much a debug tool that does almost nothing, only to ever be used by developers trying to isolate a few issues deep down :smile:

laurensvalk, 2025-05-12:
Onto something here... The StartWare APIs are often like this:

```c
    EDMA3CCPaRAMEntry paramSet;

    /* Fill the PaRAM Set with transfer specific information */
    paramSet.srcAddr = (unsigned int) buffer;
    paramSet.destAddr = UART_RBR_THR_REG;
    paramSet.aCnt = MAX_ACNT;
    paramSet.bCnt = (unsigned short) buffLength;
    paramSet.cCnt = MAX_CCNT;

    // ... and so on ...

    EDMA3SetPaRAM(SOC_EDMA30CC_0_REGS, chNum, (EDMA3CCPaRAMEntry*)&paramSet);
```

But setting is done like this deep down:

```c
void EDMA3SetPaRAM(unsigned int baseAdd, 
                   unsigned int chNum,
                   EDMA3CCPaRAMEntry* newPaRAM)
{
    unsigned int PaRAMId = chNum; /* PaRAM mapped to channel Number         */
    unsigned int i = 0;
    unsigned int *sr = (unsigned int *)newPaRAM;
    volatile unsigned int *ds;
 
    ds = (unsigned int *)(baseAdd + EDMA3CC_OPT(PaRAMId));

    for(i=0; i < EDMA3CC_PARAM_ENTRY_FIELDS; i++)
    {
        *ds = *sr;
        ds++;
        sr++;
    }
}
```

So this causes the PaRAM structs to be optimized out and the peripherals are never properly configured.

The UART echo example can at least partially be fixed with: 

```c
volatile EDMA3CCPaRAMEntry paramSet;
```

So it isn't unlikely that our SPI DMA display driver may have had the same problem.

laurensvalk, 2025-05-12:
Additionally, to get `examples/evmAM1808/cache_mmu/uartEdma_Cache.c` to work without freezing, we have to protect the following code from optimization. 

```c
#pragma GCC push_options
#pragma GCC optimize("O0")

static void config_mmu(void) {

    unsigned int index;
    /* Sets up 'Level 1" page table entries. 
     * The page table entry consists of the base address of the page
     * and the attributes for the page. The following operation is to
     * setup one-to-one mapping page table for DDR memeory range and set
     * the atributes for the same. The DDR memory range is from 0xC0000000
     * to 0xDFFFFFFF. Thus the base of the page table ranges from 0xC00 to 
     * 0xDFF. Cache(C bit) and Write Buffer(B bit) are enabled  only for
     * those page table entries which maps to DDR RAM and internal RAM.
     * All the pages in the DDR range are provided with R/W permissions
     */
    for(index = 0; index < (4*1024); index++)
    {
         if((index >= 0xC00 && index < 0xE00)|| (index == 0x800))
         {             
              pageTable[index] = (index << 20) | 0x00000C1E;
         }
         else
         {
              pageTable[index] = (index << 20) | 0x00000C12;
         }
    }
     
 
    /* Configures translation table base register
     * with pagetable base address.
     */
    CP15TtbSet((unsigned int )pageTable);

    /* Enables MMU */
    CP15MMUEnable();
 
    /* Enable Instruction Cache */
    CP15ICacheEnable();

    /* Enable Data Cache */
    CP15DCacheEnable();
}
#pragma GCC pop_options
```

So I think we now have the necessary ingredients to apply this to the SPI driver.

laurensvalk, 2025-05-12:
Is this what we call a blockbuster? I'll see myself out.

https://github.com/user-attachments/assets/154483c2-edad-41f2-8f85-bce65079a7f2

laurensvalk, 2025-05-13:
Fixed by https://github.com/pybricks/pybricks-micropython/commit/6bffa78a687e51e673458ba78bb5c09bdab5906f

Thanks everyone for the input.

closed 2025-05-13 by laurensvalk

erik-thompson-seattle, 2025-05-13:
Looks great, he just seems to be biting his lip with anxiety. Nice job.