pybricks/support #2153: [EV3 - help wanted]: Implement Pybricks USB driver

laurensvalk, 2025-04-08:
We have a temporary [USB TTY with the REPL](https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/drv/usb/usb_ev3.c) for EV3 right now. This needs to be replaced with something similar to the USB driver that was recently merged for SPIKE Prime. Then we can work towards a way to download and run programs. Initially with pybricksdev and later with Pybricks Code.

Notes:
- USB works only on the second boot. So after installing the firmware, reboot the hub.
- We may need to use a different USB library since the [Stellaris library](https://github.com/pybricks/pybricks-micropython/tree/master/lib/tiam1808/usblib) license does not seem to be compatible.

laurensvalk, 2025-05-16:
Assuming we can continue using the existing library, I suppose what we need to do is copy the [CDC driver that we currently use](https://github.com/pybricks/pybricks-micropython/blob/master/lib/tiam1808/usblib/device/usbdcdc.c) to `lib/pbio/drv/usb/ev3_usbd`, just like we have `lib/pbio/drv/usb/stm32_usbd`. Change the namespace from `CDC` to `Pybricks` and modify it to work exactly like [`stm32_usbd`](https://github.com/pybricks/pybricks-micropython/tree/master/lib/pbio/drv/usb/stm32_usbd). Equivalent descriptors, handlers, and so on.

I would gladly donate an EV3 set to the cause.

erik-thompson-seattle, 2025-05-19:
Sounds like this is the next one. I should have more time as I don't have any travel planned for the next month or so. I will start looking into this one if that's of help?

Don't need any EV3 sets. have ~30-50 units, I haven't counted.

laurensvalk, 2025-05-20:
Yes, any help here would be fantastic! Feel free to keep us posted along the way. Happy to get you up to speed or perhaps do a video call if you get stuck.

erik-thompson-seattle, 2025-05-20:
Yes, I will likely take you or @dlech up on your offers.

dlech, 2025-07-01:
> right?

Yes, sounds right.

> Couldn't VID + PID reuse conflict with existing drivers?

There is another field, `bcdDevice`, that we can use to make it look like a different version of the device so it won't cause problems with existing apps. Also the fact that we don't provide the same Class/Interface will avoid conflicts.

> I don't have an EV3 lying around

I've been experimenting with [labgrid](https://labgrid.readthedocs.io/en/latest/index.html) for work. We could potentially set that up to allow remote access for anyone that wants to poke around on it.

And if you are in the US or EU, we could potentially send you a spare EV3.

ArcaneNibble, 2025-07-05:
What level of performance are we expecting out of the USB (i.e. how close do we want to get to the theoretical 480 Mbps maximum)? Will a basic implementation be good enough? Or are we looking for a highly-optimized implementation with advanced FIFO and buffer management?

dlech, 2025-07-05:
> The license in the USB library that comes with the TI StarterWare is somewhat problematic.

Because we have other code that is GPL?

> could we hook up the low level USB peripheral to the STM32 HAL

I think that has it's own license issue:

> 4. This software or any part thereof, including modifications and/or derivative
works of this software, must be used and execute solely and exclusively on or in
combination with a microcontroller or microprocessor device manufactured by or
for STMicroelectronics.

laurensvalk, 2025-07-05:
Sounds good! Good to have that covered if we need to.

ArcaneNibble, 2025-07-05:
Is the goal of this project to entirely avoid GPL code (only "permissive" licenses), or is the goal just to avoid incompatibilities between open-source code and vendor code?

laurensvalk, 2025-07-05:
There is so much GPL code in the original EV3 firmware that it may be tricky to avoid entirely. There might also not be much benefit in doing so? It should only affect the EV3 build and all of it is open source anyway, so it shouldn't be a limitation.

So I think our best option is to make whatever we use compatible with oneother. If I understand it correctly we can still use this USB library if we package it right. (If anyone even cares for such an old product.)

ArcaneNibble, 2025-07-07:
I'm looking into this a bit more seriously now. What software are you using on the host PC to test whether or not the USB is working as expected?

dlech, 2025-07-07:
The download and run feature is implemented in `pybricksdev`.

Also, `lsusb` is useful for seeing that we have all of the descriptors correct and `chrome://usb-internals/` in a Chromium-based web browser is useful for making sure that when we get to the WebUSB implementation, that the browser will be seeing it correctly as well.

ArcaneNibble, 2025-07-07:
How do I test the REPL/stdin/stdout over the custom USB protocol (i.e. not the current CDC protocol)?

(Sorry, I'm not familiar with the UX of Pybricks the web app, since I don't own any hardware that is actually supported)

dlech, 2025-07-07:
stdin/stdout isn't working over USB yet. That is next on my TODO list.

laurensvalk, 2025-07-11:
I ran your branch @ArcaneNibble and I can see it with `lsusb`. Very nice work!

Any idea what makes the USB not work the first time after installing the firmware? (Other implementation have this too, so this is not new.)

I think some LEGO bootloaders have a way of knowing if it was just installed by checking a word in a magic address. I wonder if an extra reset could work around it or whether we really need some time powered off as during a real reset?

ArcaneNibble, 2025-07-11:
I've just pushed the latest version of my WIP code which uses the USB DMA.

I was in the process of hooking up all of the command handlers like in the STM32 codebases, but I found that attempting to call `pbsys_command` triggers a missing symbol for `pbsys_bluetooth_rx_get_free`. I don't _quite_ understand how multiplexing between BT, USB, and the UART debug console is actually supposed to work, so refactoring that would be useful for unblocking work.

The code is still missing Microsoft/WebUSB descriptors. This is not complicated, but it would be *very* useful to pull a bunch of shared code out of the STM32 implementation into a common location.

> do something similar (long term) for the at91sam7s256

This looks like it should be very possible to do in the future, yes.

> Any idea what makes the USB not work the first time after installing the firmware?

I have guesses, but I don't know for sure. My guess is that there is some kind of timing issue (interacting with host controller debouncing?) around soft-disconnect signalling.

laurensvalk, 2025-07-11:
UART i/o is for debugging only and should ultimately not be part of stdio. It's currently used with the MicroPython REPL only because there isn't something better yet. The sensor port will just be used for sensors.

I think USB and BLE are going to be duplicated, but quite a bit of refactoring is probably needed.

dlech, 2025-07-11:
> I don't _quite_ understand how multiplexing between BT, USB, and the UART debug console is actually supposed to work, so refactoring that would be useful for unblocking work.

I'll see what I can do. For the time being we can just return an error for all commands and focus on getting the descriptors right and look into the issue that Laurens reported.

And if we can keep splitting out little parts and getting them merged like like we just did with getting the Bluetooth address, that would be nice.

ArcaneNibble, 2025-07-11:
Is there a plan for what will be done with Bluetooth on the EV3? As far as I can tell, there is no BLE support on this hardware. I noticed that `lib/` has several different Bluetooth stacks as well.

dlech, 2025-07-11:
Ideally, we would implement Bluetooth on NXT first since it has the most limited (oldest) Bluetooth chip. It has an option where you just put the chip in RFCOMM mode and don't have to deal with any overhead of HCI packets.  So we may be limited to just using an RFCOMM channel because of that and end up adding a wrapper for the Pybricks Profile on that. On the PC side, we will have to make some kind of helper app. We already experimented with using RFCOMM and WebSerial but it really doesn't work well when it comes to connecting and disconnecting and the overall user experience.

But if it turns out we can do GATT over Bluetooth classic on the NXT without eating up a bunch of flash memory, we could go that way as well. But I haven't really looked into that, so I don't know if that is even remotely feasible.

On EV3, we will likely use btstack for the Bluetooth stack, but it will be different from what we have already in pbio/drv since EV3 is classic only.

ArcaneNibble, 2025-07-11:
Huh, I had no idea that GATT over Bluetooth classic even existed, but it apparently *does* work.

Are there plans to support pairing game controllers to robot bricks in order to use them as input devices?

ArcaneNibble, 2025-07-12:
> Any idea what makes the USB not work the first time after installing the firmware?

I messed around with this and adding various delays between disconnects and reconnecting, but none of them helped. I current don't have any further ideas why a full reset is required for USB to work.

dlech, 2025-07-12:
When flashing the firmware, it uses USB, so the USB peripheral is still going to be configured for that. So I suspect that there is some latent config that we aren't writing over that is causing problems. Are we completely resetting the USB peripherals during pbdrv_init()?

ArcaneNibble, 2025-07-12:
Yes, the current code does both a USB core reset (via CTRLR bit 0) as well as a USB 2.0 PHY reset (via CFGCHIP2)

ArcaneNibble, 2025-07-12:
The paragraph right before that says "When the RESET bit in the control register (CTRLR) is set, all the USB controller registers and DMA operations are reset. The bit is cleared automatically."

I think the paragraph you quoted means that resetting the *CPU core* doesn't reset the USB peripheral, whereas using the CTRLR bit does reset the peripheral.

In any case, I tried adding

```c
PSCModuleControl(SOC_PSC_1_REGS, HW_PSC_USB0, PSC_POWERDOMAIN_ALWAYS_ON, PSC_MDCTL_NEXT_DISABLE);
```

and it still didn't help.

Apparently USB high speed disconnect detection is different from FS/LS disconnect detection, and it involves removing the high-speed differential termination resistor and having the host detect a corresponding voltage rise. This is what I would *assume* the `SOFTCONN` bit is supposed to be doing, but I'm not sure why it isn't working.

ArcaneNibble, 2025-08-05:
Should we consider this "done" and open new issues for lingering pybricksdev issues?

ArcaneNibble, 2025-08-05:
The EV3 hub now implements the Pybricks USB protocol. Notable remaining issues are currently being tracked in https://github.com/pybricks/support/issues/2306

closed 2025-08-05 by ArcaneNibble