pybricks/support #2154: [EV3 - help wanted]: Implement display API

laurensvalk, 2025-04-08:
We have a [basic display driver](https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/drv/display/display_ev3.c) but right now it only shows the splash boot screen.

We'll need to implement the user API that is as compatible with [the 2.0 API](https://pybricks.com/ev3-micropython/hubs.html#pybricks.hubs.EV3Brick.screen.clear) as we can realistically do. Font handling may be a bit more basic, and file systems are not yet implemented, but we should be able to get quite close with everything else.

This API would write to a static user buffer and set `pbdrv_display_user_frame_update_requested`. Then the display driver will write it to the physical device (this is already implemented).

There are various monochrome graphics libraries we could use for this, or build something minimal of our own. I don't have a strong preference. In principle we have four colors (off, light grey, dark gray, black) but pure monochrome (on/off) is fine by me.

Related to: https://github.com/pybricks/support/issues/2152

schodet, 2025-07-05:
[image]

I would need screenshot tool to check it’s pixel perfect… or build on host.

schodet, 2025-07-05:
What is the plan:

- add a `lib/image` directory for low level C code handling drawing:
  - `lib/pbimage` would break the tab completion for `lib/pbio` :)
  - it’s reusable code not tied to pybricks.
- extend pbio to publish the display in a platform agnostic way (same way as light_matrix for example),
- add a Image class to python,
- have an Image representing the screen and use it to populate the hub `screen` object,
- use ev3dev-media project to provide “image files”,

About fonts, I think we should handle more than English language, I would like to use FreeType. I think we have enough resources on the EV3 to do it, I did it previously on a STM32 with SDRAM.

However, it would be nice to have a simple bitmap only fallback that can be used with the NXT in the future.

About image “files”, I am not sure what is your plans for the EV3. I can make it work by embedding image data in the python script for the moment.

What I have today:

- the lib/image with some drawing primitives,
- a hack in hello_world to run my code :)

My next step is to build the Image class, so that testing is easier.

laurensvalk, 2025-07-05:
This looks great! Thanks for sharing your progress update.

> add a lib/image directory for low level C code handling drawing
> lib/pbimage would break the tab completion for lib/pbio :)
> it’s reusable code not tied to pybricks.

If you're including a big existing library, then that makes sense indeed. If it's a new library you're developing for this, I think `lib/pbio/display/` would be a good place. The namespace could be `pbio_display_....` At a glance, it sounds like it could be conceptually similar to the motor PID implementation or light animations that we have in there, which is also not only for LEGO devices.

> About fonts, I think we should handle more than English language, I would like to use FreeType. I think we have enough resources on the EV3 to do it, I did it previously on a STM32 with SDRAM. [...]

Both of these make a lot of sense, and I would be glad to follow your preference.

dlech, 2025-07-05:
FWIW, I would be tempted to call it graphics (or image) rather than display since it is more about drawing that showing.

dlech, 2025-07-05:
> * sources in `lib/pbio/src/image/something.c`,
> * headers in `lib/include/pbio/something.h`,
> * function and type prefix `pbio_image_`?

This one - to be consistent with all of the other code there.

If there are "internal" headers that are only used between `lib/pbio/src/image/something.c` files, they can go in `lib/pbio/src/image/something.h`. For any "public" headers that would be used outside of the `lib/pbio/src/image/` directory, we should have preferably one header `lib/include/pbio/image.h`.

schodet, 2025-07-07:
Pushed a “work in progress”.

I am not sure where to put the python Image class, I put it temporarily in ev3devices module, but I know this is not the right place. Also not sure about the “common” directory for it. Maybe in media?

There are:
- a driver abstraction,
- the Image class,
- drawing:
  - clear,
  - sub-images,
  - single pixel,
  - image in image,
  - rectangle, filled rectangle, rounded rectangle, rounded filled rectangle,
  - circle and disc,
- python binding for all this.

Next steps:
- tests, tests, and tests,
- find a place for Image class,
- add display object in hub,
- add ev3dev media,
- implement line (easy), and thick line (hum… fun),

Then:
- text
- loading/saving from/to byte strings, or files.

Much later:
- a gui system?

dlech, 2025-07-07:
> Pushed a “work in progress”.

Great! Feel free to start a pull request for this. I would much rather get this merged in small pices rather than try to review everthing all at one time when it is "done".

> Is there a better way to run a python script?

For now, use paste mode in the REPL (CTRL+E, paste, CTRL+D). When we get USB support impelemented, then we will be able to use `pybricksdev run usb`, but that doesn't work yet.

> I will need to increase the heap, a 100x100 image does not fit.

I would _really_ rather not use the MicroPython heap for image data since it is so huge. It will make garbage collecting take a long time and lead to fragementation issues. Instead, I would like to add a second heap with some sort of alloc/free functions. Then the MicroPython image objects can use that `alloc()` to separately allocate the image buffer.  Then use a finalizer on the object to call `free()` (so create the object with `m_new_obj_with_finaliser()`.

laurensvalk, 2025-07-08:
This is fantastic @schodet ! Can't wait to give this a try and dig into the code. Now we have the good problem of trying to keep up with all code contributions 😄

laurensvalk, 2025-07-09:
It's good that we're keeping `EV3Brick.screen` backwards compatible, but maybe we can think a little bit more about where the builtin images should go and how media should work.

The implementation in the 2.0 release was very much guided by how `ev3dev` works, but that doesn't necessarily apply here.

For the SPIKE Prime Hub, we have ready-made icons in `pybricks.parameters.Icon`. It has attributes such as `Icon.UP`, but can also be used to instantiate your own Icons as `Icon(params...)`. It seems like we could do the same for `Image` and `SoundFile` too, in the existing `pybricks.parameters` module.

We could keep `media.ev3dev` as an alias for backwards compatibility.

schodet, 2025-07-13:
`pybricks.parameters.Icon` is for constant icons values, the actual parameter is a `pybricks.tools.Matrix`. So there is the same difference between a predefined image and the image type. `pybricks.parameters` doc says: “Constant parameters/arguments for the Pybricks API”. I do not think this is the right place for the Image type.

I would prefer to have the predefined images embedded in the firmware so that when you use python directly, they are available by default. However, the “file” name may not be appropriate as long as there is no file system.

How do you plan to send “files” to the brick, do you have such thing for other bricks or is it a new concept?

How compatible with pybrick 2.0 do you want to be? Do we want to accept the same strings for predefined `Image` construction?

I think supporting the `ImageFile.ACCEPT`, etc. is enough, with `ACCEPT` being some kind of opaque object, not a string like it was on pybrick 2.0. It could return an `Image` if we accept the implied memory consumption, which may be OK for EV3, but not for NXT.

Another idea: have `Image` be a full bitmap or a compressed bitmap, but invisible to the user. If you draw in a compressed image, it is first decompressed. If you use it as a source image, the code can use the compressed form directly.

I hope I am clear enough :)

laurensvalk, 2025-07-14:
> `pybricks.parameters.Icon` is for constant icons values, the actual parameter is a `pybricks.tools.Matrix`. So there is the same difference between a predefined image and the image type. [...]

They are not constants, so we can update the docstring. Icon is still a good comparison (try `Icon(12345)`), but perhaps `Color` is another example closer to home. There is `Color.RED`, but you can also make and manipulate your own colors such as `Color(h=123, s=50, v=100)`.

We also used to have a separate `geometry` module with the `Axis` class. In the end we decided to move it to parameters too so we didn't end up with so many modules.

> I would prefer to have the predefined images embedded in the firmware so that when you use python directly, they are available by default. 

Yes, this would be good. This is perhaps something for a future addition. We can bundle them with the firmware when we decide which formats we want to support.

> How compatible with pybrick 2.0 do you want to be? Do we want to accept the same strings for predefined Image construction?

It has been a little while since I've used it, and perhaps those reading this. Could you add some examples of what used to work and what you are asking we should keep? Thanks!

ArcaneNibble, 2025-07-17:
Ugh, that's indeed going to need fixing. However, REPL issues and/or IME issues technically don't block rendering.

schodet, 2025-08-02:
Suggested simple changes with ev3dev API:

- `media.ev3dev.Image -> parameters.Image`: done.
- `….Image("_screen_")`, equivalent to `brick.screen`: not needed, can we remove this?
- `media.ev3dev.ImageFile.THUMB_UP` and co: move them to `parameters.Image.THUMB_UP`
- `media.ev3dev.ImageFile.THUMB_UP` is a string, my proposal is that `parameters.Image.THUMB_UP` returns a `parameters.Image`, details are hidden to user.
- `media.ev3dev.Image("/usr/share/images/ev3dev/mono/information/thumbs_up.png")` loads a thumb up image, my proposal is to not support this.

I think we can decide this without taking the decision on the “file” format.

schodet, 2025-08-03:
About refresh: since the new event loop, the screen is updated after every draw operation, which may be slow. We need to find a good idea to avoid asking the user to be explicit about refreshes.

dlech, 2025-08-05:
> We need to find a good idea to avoid asking the user to be explicit about refreshes.

Why do we want to hide this from the user? I've always created a `Image` object, then drew whatever I wanted on it, then copied it to the screen when I was done so that the full screen updated all at once.

> So we could make it display as soon as requested but no faster than the allowed number of frames per second.

I don't think we should artificially slow down the update rate of the screen to try to automagically make things work better for users who draw directly on the screen. 

We could probably just set up a DMA triggered by a timer to get a fixed frame rate and avoid the need to use any CPU for updating the display.

laurensvalk, 2025-08-05:
> > So we could make it display as soon as requested but no faster than the allowed number of frames per second.
> 
> I don't think we should artificially slow down the update rate of the screen to try to automagically make things work better for users who draw directly on the screen.

There is not much point in updating any faster than the display can handle. "as soon as physically possible" seems reasonable to me, so when `update_requested && pbio_os_timer_is_expired(fps_timer)`

> We could probably just set up a DMA triggered by a timer to get a fixed frame rate and avoid the need to use any CPU for updating the display.

The time consuming part is likely converting the frame buffer into the format needed for the display, not the logic to start/stop transfers.

schodet, 2025-08-05:
Thanks for your answers, yes, I am also interested for the NXT!