pybricks/support #2155: [EV3 - help wanted]: Implement sound driver and API

laurensvalk, 2025-04-08:
The EV3 firmware does not have a sound driver yet. Who wants to write one?

There isn't support for a file system yet, but it would be great if the sound driver takes sound files into account.

For bonus points, are there any basic text-to-speech libraries we can use if we want to implement [the original speaker API](https://pybricks.com/ev3-micropython/hubs.html#pybricks.hubs.EV3Brick.speaker.beep) in full?

ArcaneNibble, 2025-07-14:
I wanted to check the EV3's audio PWM filter just to get an idea how much (over)engineering is warranted.

[image]

This definitely isn't very high-fidelity. It looks like playing back samples at something like 16 kHz would be plenty?

ArcaneNibble, 2025-07-16:
Minimum viable product?

https://github.com/user-attachments/assets/3dc4c79b-18c3-4b1e-9867-0755cd107663

laurensvalk, 2025-07-18:
> I was getting a bit frustrated that the only thing I seemed to be working on was "annoying" bringup

Apologies if we appear overly curt. We're very happy with all the contributions, but appear to have some trouble keeping up 😄 (This is a good problem to have!)

dlech, 2025-07-18:
> it has been ported to run on all sorts of hardware devices and displays

Including ev3dev! https://www.youtube.com/watch?v=f8I2AMFNUoY

BertLindeman, 2025-07-18:
Yep, crtl-D works, I did not even try before..

So crtl-D resarts REPL so I can use it and the video runs on and music stopped. Nice and quiet.

Thanks.

laurensvalk, 2025-07-18:
> Huh, I had no idea it actually did that. Does the code call `pbdrv_sound_stop` to stop all sounds when the REPL restarts?

You've discovered why we run MicroPython (or its REPL) as the application rather than the OS :smile:

When pressing stop, things safely stop. This is true for sounds but especially motors. With stock MicroPython, it will just drive off the table.

laurensvalk, 2025-07-24:
I think this might be mentioned somewhere, but there is an audible click on every boot, perhaps as the speaker is initialized?

BertLindeman, 2025-07-24:
> Perhaps we need our own boot jingle to mask it?

But with the possibility to set it OFF please 😀

laurensvalk, 2025-07-24:
We have an issue about a startup and shutdown jingle somewhere. Could be an internal one. I'll try digging it up.

>  it happens whenever the speaker amplifier is enabled.

What about enabling it when sound is first used?

ArcaneNibble, 2025-07-24:
You could do that, but it still generates a pop which can be heard. A previous implementation turned the amplifier on before each sound but also turned it off when the sound was done playing. We can make it turn the amplifier on once and leave it on, but I don't know if that's necessarily better?

BertLindeman, 2025-07-24:
Not sure if this is (now) the correct place to report that the NXT does not use the set volume. All at 100%

laurensvalk, 2025-08-04:
> You could do that, but it still generates a pop which can be heard. A previous implementation turned the amplifier on before each sound but also turned it off when the sound was done playing. We can make it turn the amplifier on once and leave it on, but I don't know if that's necessarily better?

Most applications don't use any sound, so deferring the pop to the first sound should mean that most users never hear it. Leaving it on after the first sound to prevent further pops sounds good to me.