pybricks/support #2156: [Bug]

merlyn383, 2025-04-11:
**Describe the bug**
What is the problem?
I cannot update the firmware on a Spike Prime hub. The Pybricks firmware was installed last summer and is currently prompting for an update. When we try to update it we get error 128. Windows 11, 24H2.

[image]

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
What did you expect to happen instead?
The firmware to be updated.

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

Full error
```
Error: unsupported hub type: 128
    at Ew (https://code.pybricks.com/static/js/main.bd3a23eb.js:2:545242)
    at https://code.pybricks.com/static/js/main.bd3a23eb.js:2:545493
    at D.<computed> (https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1381363)
    at https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1385294
    at m (https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1386225)
    at h (https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1385751)
    at q (https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1385491)
    at https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1382113
    at h (https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1378437)
    at D.<computed> (https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1382093)
```

dlech, 2025-04-11:
That's strange. 128 is the Technic hub. The firmware should have 129 to match the SPIKE Prime hub. So it's like it is trying to install the wrong firmware on the hub for some reason.

Can you share the exact steps you are doing to try to update the firmware and share which version of Pybricks Code you are using? (The version is in the About dialog.)

merlyn383, 2025-04-12:
Curious.

On the hub, hold down the bluetooth button and plug in the USB cable. Wait until the lights flash.
On https://code.pybricks.com/, click Upgrade Firmware from the left nav (MicroPython v3.6.1, Pybricks Code v2.6.0)
Select the Spike Prime Hub [cid:ad1daeee-9837-4c76-9185-26efa0152d4d]
Agree to T&C
Leave the hub name as is (Lord Robington)
Click Install

We have 2 Spike Prime Hubs, both acting the same.

Possibly related, we first tried updating the firmware on an arm64 Mac and an x86 Mac. In both cases neither computer could see the hub when it was attached via USB (it never showed up in System Report) so then we went to the gaming PC where we at least got partway and could get an actual error. Both hubs are able to connect via Bluetooth to the original Mac (arm64) and new code can be deployed, they just didn't like the USB connection. Both macs are running the latest macOS, 15.4.

Also, for the record, I did follow the instructions to manually configure the USB driver on the PC before attempting the upgrade.

If there's anything else I can provide to help troubleshoot, let me know.

-Kevin Ayers

External.

dlech, 2025-04-12:
This is very strange indeed. I just did a firmware update on both M1 Mac and Windows 11 without a problem, so I don't think there is any bug in the software that is affecting everyone. And the fact that you have two hubs that both have the same issues makes me think it might not be a hardware issue with the hub itself. So that just leaves maybe a bad USB cable?

laurensvalk, 2025-12-05:
@merlyn383 : Were you able to install the firmware in the end?