pybricks/support #2157: [Bug] "Spike Prime Works on PC but Not on Galaxy Tab S9 FE+ with Pybricks"

hashen0608, 2025-04-12:
Describe the bug
When trying to connect the LEGO Spike Prime hub using Pybricks on a Samsung Galaxy Tab S9 FE+ (model: SM-X610), the error message "No hub detected" appears. The hub is not recognized at all. However, the same hub connects successfully on a Windows PC using the same Pybricks web app.

I’ve read that the issue might be related to the Bluetooth version or compatibility on the tablet.

To reproduce
Steps to reproduce the behavior:
1. Open the Pybricks web app on Samsung Galaxy Tab S9 FE+ (SM-X610).
2. Power on the LEGO Spike Prime hub.
3. Tap the Bluetooth icon to try to connect the hub.
4. Observe the message: "No hub detected."

Expected behavior
The Pybricks app should detect and connect to the Spike Prime hub via Bluetooth, as it does on my PC.

[image]

Screenshot of the error message:

[image]

dlech, 2025-04-12:
> 1. Open the Pybricks web app on Samsung Galaxy Tab S9 FE+ (SM-X610).

What web browser are you using? Does that web browser support Web Bluetooth? We've only tested with Google Chrome on Android.

Does the browser have permission to use Bluetooth in the app settings?

hashen0608, 2025-04-17:
Hello, the issue with downloading the application has been resolved by using the Chrome browser — thank you very much!

However, I still have two questions:

When using my tablet, the response seems a bit slow compared to using a computer.

Is the drive forever block no longer usable? It used to work before, so I’m wondering if it’s related to a recent Pybricks update.

Thank you in advance for your help!

https://github.com/user-attachments/assets/f7a638f3-54c5-4566-819c-b4011b5559b7

[image]

BertLindeman, 2025-04-17:
Hi,

On the current beta.pybricks.com I see:

[image]

The about box of pybricks  shows:
```About Pybricks Beta
Pybricks logo
MicroPython for LEGO® Powered Up smart hubs.

v3.6.1 (Pybricks Beta v2.7.0-beta.1)

Copyright (c) 2020-2025 The Pybricks Authors
```

What shows the about box on your tablet?

Bert

dlech, 2025-05-19:
I'm going to close this since the original issue was solved. If you have any more questions, feel free to start a new discussion at https://github.com/orgs/pybricks/discussions

closed 2025-05-19 by dlech