pybricks/support #2158: [Bug]

DJuego, 2025-04-12:
Best regards community!

**Describe the bug**

I am working on Windows 10 and Google Chrome. This is my first time with Pybricks. I am debuting my two Spike Hub bricks (1 Spike Prime Hub + 1 MindStorms Robot Inventor Hub). 

**To reproduce**

I follow the instructions to install the Pybricks firmware on the Spike Hubs.

I successfully activate the LEGO Technic Large Hub in DFU Mode. Then I follow the instructions for installing the Windows DFU USB driver. No problem.

[image]

This is invariably the result I get. 
[image]

I have repeated the operations for several days on the two hubs I own, and with the two new cables that came with each hub. I always get the same result. 

Any ideas on what might be going on and how to resolve it?

Thank you very much!

DJuego

dlech, 2025-04-12:
Hi @DJuego, nice to see you again! 

Since you were able to update the driver and the hub shows up in the Windows Device Manager, it sounds like all of the hardware and cables are OK. So I'm not sure why it would not be showing up in the browser.

Just to be sure, if you double-click on the device, it does show that the WinUsb driver is being used?

[image]

And the USB Vendor and Product IDs match:

[image]

Maybe there is some permissions issue with the web browser not allowing access to USB devices?

DJuego, 2025-04-12:
First of all, I want you to know that I am honored that you remember me. @dlech . 

For me it was a surprise to discover that you were also behind this **Pybricks** platform. A great surprise indeed. 

I can assure you that we have been using **ev3dev** as an educational platform for years. It is for us an essential tool. Precisely on Friday of this week we had a “final” of our particular educational initiative. But this is Pybricks and I don't want to spam another project here. If you want more details just ask me.

Getting back to the subject at hand these are the screenshots that I believe answer your comments. Everything looks correct, doesn't it? There are differences, but are they significant?

[image]

[image]

Yes. I too have harbored suspicions about the browser... but we haven't touched anything and I don't think the default state is to not allow access to USB devices. After all I understand that most Pybricks users haven't had to deal with this problem, and that's why they don't mention it.

That said; it is true that we are using Google Chrome Portable and also that we have never used other browsers. Tomorrow I'm going to try to test with a standard installation of Google Chrome and with other browsers and on other computers, and see what happens.

Thanks for responding so quickly! I am delighted to cross words with you again.

DJuego

dlech, 2025-04-12:
> Everything looks correct, doesn't it? There are differences, but are they significant?

It all looks correct to me. The different product ID of 0x0011 is for the Robot Inventor hub instead of the 0x0008 I have for SPIKE Prime from my screenshot.

Is there a chance you accidentally clicked on the SPIKE Essential hub instead of one of the other two?

DJuego, 2025-04-13:
Well. I tried Firefox Portable.

[image]

This hint about that Web USB deserves investigation on my part. Maybe I have it deactivated (?) I don't know anything about this Web USB.

Next I tried Microsoft Edge (native installation) and got the same result.

[image]

I am going to try again with my laptop although I have already tried it a few days ago with identical results.

DJuego

PS: Of course I follow the DFU mode instructions. On that there are no problems. The led flashes pink, green, blue and off.

dlech, 2025-04-13:
> Firefox

Firefox and many other browsers don't support Web Bluetooth and Web USB. On Windows there is just Chrome and Edge that support it. And they are basically the same in this regard (based on the same code base).

> I am going to try again with my laptop

Are these school computers, I wonder if it is possible if the IT department has something locked down.

---

Some more things to check:

Enter `chrome://settings/content/usbDevices` in the address bar and make sure permission is granted for USB. (`chrome` will redirect to `edge` for that browser).

[image]

And there is also `chrome://usb-internals/` where the device should show up. You may need to refresh this page if you connect USB after opening the page.

[image]

If the hub shows up here, then there could be a problem with Pybricks Code. If the hub doesn't show up here, then there is probably something wrong/misconfigured on the computer.

DJuego, 2025-04-13:
Thanks for all those valuable details, @dlech . Indeed; they can be really useful for me and others.

> Are these school computers, I wonder if it is possible if the IT department has something locked down.

In this case I am working with my personal desktop PC  and a laptop that I manage in its entirety. :-)

Updates!

I test with  MindStorms Robot Inventor. I have:

[image]

> Enter chrome://settings/content/usbDevices in the address bar and make sure permission is granted for USB. (chrome will redirect to edge for that browser).

[image]

> And there is also chrome://usb-internals/ where the device should show up. You may need to refresh this page if you connect USB after opening the page.

[image]

NOTE:  These are the same connections that are seen when the Hub is not connected to the PC, so none of these connections have anything to do with the Hub.  Yes, that's right. I have FIVE disks connected to my computer (crazy!), the keyboard and mouse are USB. I even have a Bluetooth dongle connected because my PC does not have Bluetooth built in (corresponds in the table to port number 12). Port 3 in the table I don't know which device it corresponds to, but I think it is irrelevant to my problem.

HOWEVER. I have tested with the laptop, also with Google Chrome Portable... and it works! With Spike Prime Hub AND MindStorms Robot Inventor.

[image]

I developed a small program that turned on a motor when the force sensor was pressed. It worked perfectly. Cool!

CONCLUSION: The problem is on my desktop computer. maybe too many USB devices connected at the same time? My turn to look into it but don't worry. The main problem I consider it solved now. Thank you very much for your hints! 

I ask for a few days before closing this thread in case I find out what the problem is between Pybricks and my desktop PC.

DJuego

PS: Some additional non-bug related comments:

I was surprised that to program I had to connect via Bluetooth even though the USB was already connected. I'm afraid it's going to take me a while to get used to these Hubs and Pybricks. So far the Mindstorms EV3s seem vastly superior to me, even if they don't have a cool LED array. I feel it as a step backwards from LEGO. And on top of that at the same or higher price than the Mindstorms EV3 was when it went on sale. :-) I imagine as I interact with it I'll start to see the advantages. 

At the moment I miss the linux environment that offered me more versatility, although I don't think it's bad if you can get much more power autonomy by slimming down the hardware.

I recognize that it is much lighter and smaller, which is advantageous... but I don't like at all that the batteries can only be charged in a HUB. I don't like the loss of interoperability with the motors and sensors of the previous kit although I understand it and there is an unofficial adapter. I don't like that an OpenMV camera can't be connected directly although I understand it and there is also an unofficial adapter. However it does not have any Wifi support that I know of. :-((

jcarlos-ciber, 2025-05-27:
Indeed. I confirm that the problem was mysteriously corrected. I couldn't identify what was causing it, but it probably had more to do with my desktop and laptop than with Pybricks. 

The firmware  on both my bricks was successfully updated. When I have new bricks to update we will see if it happens again.

In the meantime I request that this issue be closed. Thanks!

DJuego

DJuego, 2025-05-27:
Indeed. I confirm that the problem was mysteriously corrected. I couldn't identify what was causing it, but it probably had more to do with my desktop and laptop than with Pybricks.

The firmware on both my bricks was successfully updated. When I have new bricks to update we will see if it happens again.

In the meantime I request that this issue be closed. Thanks!

DJuego

closed 2025-05-27 by DJuego