pybricks/support #2176: [Bug] city hub re-powers ON after button-keep to power-off on github hash 191b715d and older

BertLindeman, 2025-04-30:
Sorry for the long post, but I want to be complete. . . 

**Describe the bug**
cityhub at `ci-build-3813-v3.6.1-152-g191b715d' after power-off (with the hub button) the hub restarts

Have NOT noticed this on movehub, Technichub, primehub nor Robot Inventor hub at the same firmware.

Tested a number of firmwares for the cityhub.
Not all of them from the master branch.
The oldest I tried was cityhub-firmware-build-3776-git91d7221e
Same result.

To be sure it is no hardware thing I flashed the current beta.
No problem at that firmware - ci-release-86-v3.6.1 on 2025-03-11 

**To reproduce 1**
Steps to reproduce the behavior:
1. Put the hub on
2. connect to a pc
3. disconnect (using the pc button)
4. press and keep the hub button until the hub flashes fast and goes off
5. See the error: after less than a second hub restarts

**To reproduce 2**
Steps to reproduce the behavior:
1. Put the hub on
2. connect to a pc
3. run this:
```python
from pybricks.hubs import ThisHub
hub = ThisHub()
hub.system.shutdown()
```
4. hub goes off
5. hub restarts.

**Scenario difference** between hash 0663e76d and hash da108b30
This scenario does **not** restart on github hash 0663e76d.
But **does restart** at github hash da108b30:
1. pull the batteries from the hub and reinsert
2. press the hub button
3. hub starts
4. press and keep the hub button until the hub flashes and goes off
And no restart on github hash 0663e76d
But a restart on github hash da108b30

Another **no restart** scenario:
1. connect to the hub
2. enter the REPL
3. issue the command `hub.system.shutdown()`
4. hub goes off and stays off

**Expected behavior**
Hub should stay off

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

zus2, 2025-04-30:
> BertLindeman
>  created an issue
> (pybricks/support#2176)
> Sorry for the long post, but I want to be complete. . .
>
> Describe the bug
> cityhub at `ci-build-3813-v3.6.1-152-g191b715d' after power-off (with the hub button) the hub restarts
>
> Have NOT noticed this on movehub, Technichub, primehub nor Robot Inventor hub at the same firmware.
>
> Tested a number of firmwares for the cityhub.
> Not all of them from the master branch.
> The oldest I tried was cityhub-firmware-build-3776-git91d7221e
> Same result.
>
> To be sure it is no hardware thing I flashed the current beta.
> No problem at that firmware - ci-release-86-v3.6.1 on 2025-03-11
>
> To reproduce 1
> Steps to reproduce the behavior:
>
> Put the hub on
> connect to a pc
> disconnect (using the pc button)
> press and keep the hub button until the hub flashes fast and goes off
> See the error: after less than a second hub restarts
> To reproduce 2
> Steps to reproduce the behavior:
>
> Put the hub on
> connect to a pc
> run this:
> from pybricks.hubs import ThisHub
> hub = ThisHub()
> hub.system.shutdown()
> hub goes off
> hub restarts.
> Scenario difference between hash 0663e76d and hash da108b30
> This scenario does not restart on github hash 0663e76d.
> But does restart at github hash da108b30:
>
> pull the batteries from the hub and reinsert
> press the hub button
> hub starts
> press and keep the hub button until the hub flashes and goes off
> And no restart on github hash 0663e76d
> But a restart on github hash da108b30
> Another no restart scenario:
>
> connect to the hub
> enter the REPL
> issue the command hub.system.shutdown()
> hub goes off and stays off
> Expected behavior
> Hub should stay off
>
> Screenshots
> There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

Can only be a hub firmware call to reboot?  Do not know how these flags work are set or stored ,  but are used when you install / upgrade computers OSs to reboot etc.

dlech, 2025-04-30:
This is a bug we have fixed several time already. 😄  https://github.com/pybricks/support/issues/692 https://github.com/pybricks/support/issues/385

Sounds like we've managed to regress it again.

laurensvalk, 2025-05-01:
Sorry about that. There have been extensive changes to this part of the code lately. Admittedly, it hasn't been tested much on City Hub. Fortunately @BertLindeman is here to help us, and keep us on our toes. 😄

BertLindeman, 2025-05-01:
Trying to be complete (hum)

No devices attached to cityhub.
Battery: Voltage: 7599

BertLindeman, 2025-09-27:
Similar situation seen at [Commit 0f065bb](https://github.com/pybricks/pybricks-micropython/commit/0f065bbd5e87c38b148abc1e822e21cfccc5ac47)

Not verified at what exact commit this started again.
This was the last commit applied to master.

### Scenario 2 does NOT trigger the re-power on.
Scenario 2 run shutdown program:
```python
from pybricks.hubs import ThisHub
hub = ThisHub()
hub.system.shutdown()
```
Correctly powers-off and does not automatically re-powers-ON.

laurensvalk, 2025-10-06:
If that is successful, then here is a version that _should_ work the same. But this is in a cleaned up form that we could actually merge. https://github.com/pybricks/pybricks-micropython/pull/393

laurensvalk, 2025-10-07:
> Correct, but I'll try again. Might the timing a a factor? (How fast / slow I press / release the button?

I will make a firmware variation to help rule this out later today.

laurensvalk, 2025-10-07:
On second thought, this is a bit complicated to debug without being able to reproduce it.

BertLindeman, 2025-10-07:
On `('cityhub', '4.0.0a0', 'ci-build-4263-v3.6.1-521-g503b5246 on 2025-10-07')` I see no problem anymore.

Tried connected / not connected / being connected
while running a program
simply idle
directly after connected
after running a program / connected / not connected.
a few times.
Feels OK although a negative check is always difficult.

closed 2025-10-07 by laurensvalk