pybricks/support #2189: [Bug] OSError: [Errno1] EPERM: Whenever motor is under load

RitchieP, 2025-05-14:
**Describe the bug**
Currently, I have a program that would control SPIKE Prime medium motors. And recently, my hub occasionally quits running the program halfway. And shows me this error.

```
The requested operation is not valid in the current state:
--> Check the documentation for required conditions.
--> Check the line in your script that matches
    the line number given in the 'Traceback' above.
```

After a couple of observations, it seems like the program has a higher probability of quitting whenever the medium motors are under load. And after the error, if I rerun the program again, sometimes it takes a while before the program actually runs.

I've also tested the same program with another set of hub and motors, and it does not appear, so I think it's definitely not a program problem.

**To reproduce**
I'm not particularly sure how to reproduce the error; it happens kinda randomly.

Would love to hear input on how to fix this or is this a hardware problem? And also what does the error message mean by current state?

Thanks!

dlech, 2025-05-14:
If a different hub and motors works, then it is likely a wiring issue. If there is a faulty wire on the data lines, it would explain the error. The hub would think that the motor became disconnected and raise an error like this.

The "current state" in this case would possibly be the motor is disconnected. The Powered Up hubs have limited flash memory, so we can't always make a more specific error message unfortunately.

RitchieP, 2025-05-15:
Ah, I see, I'll test it out later today when I have time, and be sure to provide feedback here.

RitchieP, 2025-05-16:
Just tested it, and yes, it was a faulty wire issue. Thanks for the help!

closed 2025-05-16 by RitchieP