pybricks/support #2191: The CityHub occasionally freezes during reconnect with the Remote Control - when at the fringes of operating range

TrainsWithLights, 2025-05-15:
Describe the bug:

Occasionally, the Hub will lock up and freeze when the remote control handset is at the limit of range and trying to reconnect as a result of power up.

This scenario occurs when a Lego train on a long track (typically outdoors) travels out-of-range and causes the handset to turn off.

With the track 'loop' then returns the train to within range, the handset can be repowered to regain control of the train. But occasionally, the hub/remote handshake fails and the Hub freezes.

To reproduce:

Power up the hub, connect to the web portal and run the code below.
Power up the remote-control and the remote's name should be displayed ('Handset' is normally what is displayed)
and the remote-control's LED then starts flashing Red and Yellow.

You can confirm the code supports the reconnection just by depowering and repowering the remote control at the bench - the Red/Yellow flashing resumes. When full signal strength is available it all works well and recovers 99% of the time!

Then, with remote-control blinking Red/Yellow and in hand, just walk away from the hub.
At some distance, the remote-control's LED will start flashing White/Red/Yellow, but keep walking until just White/OFF are blinking.

At this point wait for the remote-control to power off completely.
Now re-power the remote-control and slowly start walking towards the Hub.

If the lights resume flashing Red/Yellow, the recovery has worked, so turn around and walk out-of-range again to repeat. You may need to walk back and forth once or twice before the recovery fails, but it will!

Expected behavior
The remote control should stop 'flashing white/off' and return to blinking Red/Yellow.

The error log:
[image]
The hub freezes during the error message.

The code:
```
from pybricks import version 
from pybricks.iodevices import PUPDevice
from pybricks.parameters import Button, Color, Direction, Port
from uerrno import ENODEV
from pybricks.hubs import CityHub
from pybricks.pupdevices import Remote
from pybricks.tools import wait

def Remote_LED(value):  
    try:
        my_remote.light.on(value)
  
    except OSError:                           
        print("Remote Control -> Out of range")
        Recover_Remote_Control()

def Recover_Remote_Control():

    global my_remote
    global _Remote_Name

    while True:
        try:
            print("Disconnect")
            my_remote.disconnect()
        except OSError:
            print("D-Error")

        try:
            print("Reconnecting")
            my_remote = Remote(name=_Remote_Name, timeout=2000)

            print("\n", "Remote Control -> ACTIVE", "\n", "name = ", _Remote_Name)          
            return
        except OSError as error:
            print("OSerror - ", OSError, error.errno)

print ("\n", version)

my_remote = Remote()
_Remote_Name = my_remote.name()
print("\n", "Remote Control -> ACTIVE", "\n", "name = ", _Remote_Name, "\n" ) 

while True:
    Remote_LED (Color.RED)
    wait (500)
    Remote_LED (Color.YELLOW)
    wait (500)
```

Previous discussions with the maintainers "it sounds like it would be best if we just made the remote automatically reconnect so that users don't have to do that in their program."

(Would also be very pleased if the remote control handset could be coded to stay alive for say 2 minutes rather than 12 seconds. OR does that require capacitor / resistor changes for the watchdog timer?)

dlech, 2025-05-15:
Looks like duplicate of #2190

TrainsWithLights, 2025-05-15:
Can you please delete 2190

dlech, 2025-05-15:
I already set the labels and fixed fixed up the formatting on the other issue, so I would rather close this one.

TrainsWithLights, 2025-05-25:
OK, re-opened 2190 and will now close this one soon
Any idea how long the bug-fix might take?

closed 2025-05-29 by TrainsWithLights