pybricks/support #2201: [Bug] can't upload a file backup file created on my windows pc to my Mac

Trohorm, 2025-05-18:
That's all it gives when I try to upload
```
jv@https://code.pybricks.com/static/js/main.bd3a23eb.js:2:505532
h@https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1385695
l@https://code.pybricks.com/static/js/main.bd3a23eb.js:2:1386009
```

Trohorm, 2025-05-19:
Okay so my bad, it isnt just files from my windows anything that i try to upload just fails even if it was downloaded from that computer. It just says unsopported file type text/x-python-script and the error messages in the previos post

dlech, 2025-05-19:
So the same error as https://github.com/pybricks/support/issues/1436?

This likely means that your browser isn't supporting the File System Access API. 

What browser and version are you using and which operating system?

I used to know how to work around this, but I forgot how. There is (or at least used to be) some setting in the browser to tell it to allow that file type to be downloaded. I can't find the setting or any old issues referencing this though.

Trohorm, 2025-05-19:
Yeah your right its just safari not letting me do it so i just used Chrome and it works perfectly fine. Thx for the help tho

dlech, 2025-05-19:
OK, great. Safari doesn't support Web Bluetooth either, so doesn't work for Pybricks Code. So we can close this issue.

closed 2025-05-19 by dlech