pybricks/support #2206: Technic Hub freezes when using ble.broadcast()

akiyuki-k, 2025-05-19:
I really appreciate all the great work on Pybricks.

I'm currently facing an issue where the Technic Hub freezes when calling Bluetooth functions, and I haven't been able to resolve it on my own. I would greatly appreciate any help or guidance.

The program I am using is a simple Bluetooth send-and-receive loop. It includes a receive operation, but in my testing, no messages are ever received (the received message is always empty).

Environment where the issue is reproducible:

 • Pybricks v3.6.1 (Pybricks Code v2.6.0)
 • One Technic Hub with nothing connected to any ports
 • The following program is transferred to the Hub
 • After unpairing the Bluetooth connection between the programming PC and the Hub, the program is executed → the Hub freezes
 • This occurs consistently: 20 out of 20 attempts result in a freeze
 • The time to freeze varies between 1 minute and 150 minutes

When the Hub freezes, the status light remains yellow. Based on this, I suspect the issue is related to the hub.ble.broadcast(senddata) call.

As a reference, when I keep the Bluetooth pairing between the programming PC and the Hub, the same program runs for more than 10 hours without freezing (confirmed twice).

Program:
```
from pybricks.hubs import TechnicHub
from pybricks.pupdevices import Motor
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch
from umath import pi, sin, cos, radians

hub = TechnicHub(broadcast_channel=1, observe_channels=[2])

# Receive a message every 100ms. Send a message every 1s.
# The indicator switches to red when receiving and yellow when sending.
count = 0
while True:
    hub.light.on(Color.RED)
    data = hub.ble.observe(2)
    if count > 10:
        hub.light.on(Color.YELLOW)
        senddata = ('send', 1)
        hub.ble.broadcast(senddata)
        count = 0
    wait(100)
    count += 1
```

Please let me know if there's anything else I can try or provide to help diagnose the issue. Thank you in advance!

dlech, 2025-05-19:
Do you actually need to broadcast and observe on the same hub? If you can do only one or the other, the Bluetooth chip will be happier.

Other than that, I think we need to dust of the parts of https://github.com/pybricks/pybricks-micropython/pull/80 where we transparently don't broadcast and observe at the same time on the Bluetooth chip and have some random timing to avoid aliasing issues.

akiyuki-k, 2025-05-20:
@dlech 
Thank you for your reply.
I want to run multiple hubs in sync, so I need some kind of two-way communication.

I had not yet looked at that #80 issue. I will take a look.
I understand that I need to separate sending and receiving. Which library should I use if I want to control Bluetooth in such a detailed manner?

akiyuki-k, 2025-05-21:
@dlech 
I see, so that's something that can only be handled on the firmware side. That’s not something I can try on my own.
As a request for a future firmware version, I would really appreciate it if it could support both sending and receiving.
For now, I’ll try to see if I can achieve my goal by structuring things so that it only does either sending or receiving.

Thank you very much.

laurensvalk, 2025-06-06:
Thank you @akiyuki-k for filing this issue. I'm a big fan of your work!

Even though we've looked at this at various times, we'd still like to find a better solution or workaround than we have today. I guess you have inspired us to take another look 😄 . For context, a somewhat recent attempt is https://github.com/pybricks/pybricks-micropython/pull/269, but it wasn't quite satisfactory.

> As a reference, when I keep the Bluetooth pairing between the programming PC and the Hub, the same program runs for more than 10 hours without freezing (confirmed twice).

This is really interesting and got me curious. Could you try and repeat this without the PC connected, but having `remote = Remote()` to keep the remote connected during your experiment? Not ideal perhaps, but possibly an interesting workaround.

> One Technic Hub with nothing connected to any ports

Have you been able to reproduce it on City Hub? I am able to reproduce it with your example on the Technic Hub but my City Hub has now been running for several hours without getting stuck. Even though its Bluetooth chip is the same, it has a different firmware version.

> something that can only be handled on the firmware side.

Due to recent unrelated improvements related to debugging, we now have the ability to analyse the problem in a bit more detail. There does seem to be a specific pattern of internal Bluetooth operations that always shows up when it gets stuck. If that turns out to be true, maybe there is a way we can work around it. Stay tuned...

BertLindeman, 2025-06-08:
Ran the [original program from this issue](https://github.com/pybricks/support/issues/2206#issue-3075147359)

All tests:
- no PC connection
- no attatchments

Test on a technichub (My HubB) at version `('technichub', '3.6.0b4', 'ci-release-83-v3.6.0b4 on 2025-02-14')`
Results in a hang after ~10 minutes status: Yellow. 
Hub powered down by long button press. 

Another hub (HubA) with the program below (included as Laurens suggested: `remote = Remote()`)
on firmware from build-3821 at git hash dfbcfa8. 
Still running.

<details><summary>The adapted program:</summary>

```python

from pybricks.hubs import TechnicHub
from pybricks.pupdevices import Motor, Remote
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch
from umath import pi, sin, cos, radians
C1 = 1
C2 = 2
hub = TechnicHub(broadcast_channel=C1, observe_channels=[C2])
remote = Remote()
# Receive a message every 100ms. Send a message every 1s.
# The indicator switches to red when receiving and yellow when sending.
count = 0
while True:
    hub.light.on(Color.RED)
    data = hub.ble.observe(C2)
    if data:
        hub.light.on(Color.VIOLET)
    if count > 10:
        hub.light.on(Color.YELLOW)
        senddata = ('send', 1)
        hub.ble.broadcast(senddata)
        count = 0
    wait(100)
    count += 1
```
</details>

Restarted HubB with this program too. (I have two remotes :smile: )
Both tests still running ~ half an hour now.

Will report later (Hope all batteries last long enough)

Bert

laurensvalk, 2025-06-08:
@BertLindeman , would you like to try the original on CityHub also? Thanks!

akiyuki-k, 2025-06-08:
@laurensvalk
I'm glad to hear you're a fan of my work.
I'm very grateful for everything Pybricks has done for me. thank you for looking into this issue.
The current version of Pybricks is excellent and I have no complaints, but if this issue is resolved, I believe it will allow us to build even more reliable and large-scale systems.
I'll check with my City Hub as well.

@BertLindeman
Thank you for checking. It does seem like remote = Remote() could have an impact.

BertLindeman, 2025-06-10:
This morning status after 15 hours runtime: technichubs - no problem doing red/green

Maybe related to the reset loop yesterday:
- **after** I stopped the primehub its bluetooth button was OFF and yet the technichubs recieved the data.
Not sure if the BT button was blue while the primehub was running. 
(For the night I covered the hubs to dampen the lights)
Switched the primehub off and on and set the bluetooth button to blue-blink.
The technichubs showed violet / yellow while the primehub was OFF and they pickedup red/green after the primehub restarted.

Plan: switch the 1000 msec waittime down to 250 msec and see where that leads us.

BertLindeman, 2025-06-11:
Bad news.
This testrun used `stdwait = 5`

Between 8 and 11:30 hours runtime one of the technic hubs (HubB) hang at violet.
```('technichub', '3.6.1', 'ci-build-3827-v3.6.1-161-g91b566fe on 2025-06-09')``` 

```Battery voltage: 7298 mV```

Restart of the broadcast primehub did not have effect on HubB.
HubA still runs happily; now 16 hours.

BertLindeman, 2025-06-11:
Restarted the same test with HubB  and now after two hours this hub hangs on yellow, so during the reset action.
HubA still runs on. for 18 hours.
To look inside the hubs for some identification, I stop hubA test (battery is getting low too).

HubA:

[image]

HubB:

[image]

laurensvalk, 2025-06-12:
Thanks, that's awesome @BertLindeman !

Indeed, fundamentally this is still the same method, so it can still lock up eventually. But by drastically reducing the amount of observe-restarts, it looks like we can get up times of 2 to 18 hours instead of just quarters of hours or less.

This would be a big step up, and probably as good as it gets. Multiple-hour runs would be great for folks like @akiyuki-k and @JJHackimoto when making videos.

This also makes me wonder if we ever managed to get it much better than that by alternating observe/broadcast as tried by @nstrijbosch. 

----------------------------

Since this is looking good, I think the next step is to find a good way to integrate this as a solution. Either something automatic or something that the user can do.

BertLindeman, 2025-06-12:
For the record, no idea if it helps.

Reloaded the batteries and started the test again with `stdwait = 10` with the two technic hubs.
They were idle for the moment, so no hard work.
If a hang occurs I'll report and if not I'll report tomorrow after more than 24 hours and keep the test running until battry-low.

laurensvalk, 2025-06-12:
Everyone: [Here](https://github.com/pybricks/pybricks-micropython/actions/runs/15610346897/artifacts/3314048436) is a first attempt at a builtin solution that reduces the number of restarts in the background. You can easily install it with [these instructions](https://pybricks.com/learn/getting-started/install-pybricks/#trying-the-nightly-version-optional).

It may still hang eventually, but it improves runtime significantly. In contrast to the debug versions above, _no action is required from the user_ here.

So you should be able to use this as a drop-in replacement to test it in your current setup. (In addition to other unrelated changes since the last stable release. Please report other issues separately.)

BertLindeman, 2025-06-12:
You forget to define the watch
```
watch = StopWatch()
``` in the sample program.

Similar setup here HubA with 10msec stdwait  now running for 6:20:56.
HubB new firmware and running you latest sample program.

It is very easy to see the hub still has a PC connection as I see a lot of Violet. So let me drop that connecting ;-).

laurensvalk, 2025-06-12:
The stopwatch (or any user timing) don't need to be used here. I've updated the example.

You should be able to run any of your favorite examples or GBC master pieces without changes and just see it work for much longer :wink:

BertLindeman, 2025-06-14:
Final message about the test HubA starts to blink-blink orange so  battery near low. None of the hubs suffered a hang.
Duration (also of the batteries) nearly 54 hours s 2 days 6 hours.

laurensvalk, 2025-06-14:
Was this with the new firmware, without any user restarts? That's amazing! Thanks for running them for so long.

laurensvalk, 2025-06-14:
It hasn't been merged to master yet, but you can try it out with the version linked [here](https://github.com/pybricks/support/issues/2206#issuecomment-2966642661).

The issue is technically not fixed (we think the issue is in the Bluetooth chip that we can't change), but it should be much reduced because the likelihood of certain operations clashing is much reduced.

laurensvalk, 2025-06-14:
And thank you @Debenben for sponsoring the project again so generously today!

laurensvalk, 2025-06-16:
The proposed improvement has been merged to master. You can download and install it using [these instructions](https://pybricks.com/learn/getting-started/install-pybricks/#trying-the-nightly-version-optional).

Happy to consider further improvements. Please share your findings on whether this is any better (or worse). Thanks!

laurensvalk, 2025-06-27:
Thanks for the details! This is really useful. Am I understanding correctly that you feel it improved slightly but perhaps not for all programs?

laurensvalk, 2025-06-28:
Oh, if you're running your test program (or that of OP) standalone, it is expected that there is _no improvement_. If it is getting no data, it resets it every 10 seconds just as it did before.

It's just that when it _does_ get data, it knows that it needs not reset just yet, delaying the problematic situation. There are a few examples above you can use to test, or use it in your own application alongside your other hub as you normally would.

BertLindeman, 2025-06-29:
Tested with @Debenben previous post program and have a primehub broadcast.

Hub at firmware ```('technichub', '3.6.1', 'ci-build-3854-v3.6.1-169-geb3e2e64 on 2025-06-28')```
Hub is HubB with clip bottom.

<details><summary>primehub broadcast program</summary>

```python
from pybricks.hubs import PrimeHub
from pybricks.parameters import Color, Port
from pybricks.tools import wait

hub = PrimeHub(broadcast_channel=2)
hub.display.off()
while True:
    for i in range(100):
        hub.ble.broadcast("red" if i % 2 else "green")
        hub.light.on(Color.RED if i % 2 else Color.GREEN)
        wait(250)
```

</details>

In seconds the hub stops at red or white.
<details><summary>
Because it is not so easy to see if the hub got stuck, I have a cityhub listening to the TechnicHub:</summary>

```python

from pybricks.hubs import CityHub
from pybricks.parameters import Color
from pybricks.tools import wait, StopWatch
# from urandom import random

hub = CityHub(observe_channels=[1])
watch = StopWatch()
waittime = 100
prev_data = -1
while True:
    data = hub.ble.observe(1)
    if data and data != prev_data:
        prev_data = data
        hub.light.on(Color.GREEN)
        last_data = watch.time()
        print(f"last incoming data at {last_data / 1000:7.0f} sec data: {data:4.2f}")
    else:
        if prev_data == -1:
            watch.reset()  # no data received yet
        hub.light.on(Color.YELLOW)
    wait(waittime)
``` 

</details>

So if the hub hangs at RED or WHITE, it is in the enable or disable of observe.

Removed the `observe_enable` from the program like this:
```python

from pybricks.hubs import TechnicHub
from pybricks.parameters import Color
from pybricks.tools import wait, StopWatch
from urandom import random

hub = TechnicHub(broadcast_channel=1, observe_channels=[2])
while True:
    hub.light.on(Color.RED)
    hub.ble.broadcast(random())
    hub.light.on(Color.GREEN)
    hub.light.on(Color.BLUE)
    data = hub.ble.observe(2)
    hub.light.on(Color.WHITE)
```

So no wait and as fast as possible broadcast and observe.
The GREEN and WHITE color could be taken out too, I did not.
Now have time enough to disconnect the hub ;-)

<details><summary>
Log of the CityHub shows that most of the time 4 different observations per second come in:</summary>

```

last incoming data at     598 sec data: 0.74
last incoming data at     599 sec data: 0.41
last incoming data at     599 sec data: 0.94
last incoming data at     599 sec data: 0.68
last incoming data at     600 sec data: 0.36
last incoming data at     600 sec data: 0.34
last incoming data at     600 sec data: 0.33
last incoming data at     600 sec data: 0.40
last incoming data at     601 sec data: 0.52
last incoming data at     601 sec data: 0.74
last incoming data at     601 sec data: 0.57
last incoming data at     601 sec data: 0.72
last incoming data at     601 sec data: 0.02
last incoming data at     602 sec data: 0.93
last incoming data at     602 sec data: 0.12
last incoming data at     602 sec data: 0.48
last incoming data at     602 sec data: 0.79
last incoming data at     603 sec data: 0.35
last incoming data at     603 sec data: 0.84
last incoming data at     603 sec data: 0.32
last incoming data at     603 sec data: 0.31
last incoming data at     603 sec data: 0.77
last incoming data at     604 sec data: 0.60
```
</details>

Test running on . . .