pybricks/support #2223: [Bug] Infraredsensor non lego

20100Lego, 2025-06-05:
**Describe the bug**
What is the problem?
I'm doing some advanced programming with Pybrick (thanks to you!) like a GBC module that writes text with the bullets.
I saw that the infrared sensor has a class for measuring brightness. I bought a non-Lego infrared sensor (Aliexpress), it works as a distance sensor in the poweredup application. In pybricks if I do sensor = InfraredSensor (Port.A), the script blocks and doesn't go any further, it doesn't output as an error. 
Do you think it's possible to integrate these sensors to make them pybricks compatible?
I'm a computer scientist, I can do tests to determine what the sensor returns on initialisation if you provide me with debugging/testing tools.
It would be great if it worked!
Thanks in advance!

**To reproduce**
Steps to reproduce the behavior:
1. Connect an unofficial Infraredsensor sensor
2. Try to connect sensor = InfraredSensor (Port.A
3. Script does not go ahead

**Expected behavior**
What did you expect to happen instead?
Sensor sends value

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problem

laurensvalk, 2025-06-06:
> I bought a non-Lego infrared sensor (Aliexpress) (...) if I do sensor = InfraredSensor (Port.A), the script blocks and doesn't go any further, it doesn't output as an error.

It is likely that this sensor doesn't implement the LEGO communication protocol quite right.

> Do you think it's possible to integrate these sensors to make them pybricks compatible?

Probably. We've been busy enough getting all the official stuff to work, so we haven't gotten around to some of the third party devices. 

> I'm a computer scientist, I can do tests to determine what the sensor returns on initialisation if you provide me with debugging/testing tools.

Where you ordered the sensor, you can probably also order a spare cable that you can cut and wire up to a TTL/USB cable. Combined with a custom built firmware, you will then be able to monitor exactly where it gets stuck.

If this sounds like a fun challenge, let me know and we can provide some instructions 😄

20100Lego, 2025-06-06:
Thank you for your answers!
I'll try to test the other classes this weekend.

I realised that Lego no longer sold colour sensors, so I wanted to try a third-party sensor.

Yes, I can imagine that it's already quite a job to implement the whole Lego suite.
If I have to analyse and provide you with the information you need to develop pybrick, I'm all for it!

I ordered a female pup plug and a usb ttl adapter, otherwise I must have an Arduino somewhere.
If you can tell me how you analyse what the sensor sends back, that would be great.
Thanks a lot

laurensvalk, 2025-06-06:
Since you've looked into it, do you know if there is one major party producing such sensors or is it more likely a mix with potentially lots of different and incompatible versions?

nuno2500, 2025-07-23:
Hi,

Same problem with me. Thaks ok. When I bought the sensor, I already knew that maybe wouldn't work and could be used in another way. Is it possible to use P1 to P4 to power any motor, LED, or even an electromagnet? The maximum amperage and voltage could be limited to avoid frying the hub. What do you think?

Best Reagards,

Nuno Caravaca

Portugal

nuno2500, 2025-07-23:
> There is some info on https://bricks.stackexchange.com that shows what is inside the hubs, for example for the City Hub: https://bricks.stackexchange.com/a/10258/3498
> 
> Here you can see the motor controller chip and there is also a fuse that limits the current from the battery. So you can us this info to determine if it will work with whatever electronics you dream up.
> 
> But in general, yes you could hook up anything that can run off of battery power to P1 and P2. You just have to be careful about the polarity.

Thanks for your tip but how can i send voltage to P1 and P2 with PyBricks code? I do use coding by watching the exemples in the PB docs.

Best Regards

Nuno

nuno2500, 2025-10-20:
Done.