pybricks/support #2227: [Feature] A 'double stud' run external task block, please

antonvh, 2025-06-09:
**Is your feature request related to a problem? Please describe.**
I want to import an external **init** task into block code, where I call pd = PUPDevice(Port.A)
When the block code switches to async, because I have multiple start blocks, or a SubTask, it refuses to execute the imported init task, because PUPDevices cannon be instantiated async.

**Describe the solution you'd like**
If I were able to run the imported **init** routine in the setup stack, that would solve it. The other routines can be made async.

**Describe alternatives you've considered**
- Converting the block code project to pure python, but I love blocks!
- some workaround using `import *` - dirty.

**Additional context**
I expect the double stud version here:

[image]

laurensvalk, 2025-06-13:
This would be nice. You can also do it with existing blocks:

[image]

antonvh, 2025-06-14:
> Related requests for improvement: #1635

Thanks a lot. This is really helpful!