pybricks/support #2230: `ubluetooth` support (at least for Prime Hubs)

romw314, 2025-06-11:
> We won't be enabling `ubluetooth`, but we're looking at a simple but effective serial stream over BLE. That's the aim of the V3.3 release.

 _Originally posted by @laurensvalk in [#140](https://github.com/pybricks/support/issues/140#issuecomment-889827027)_

Why you won't be enabling `ubluetooth`? It is hard to enable it? Because I want to transfer lots of data not possible with Pybricks BLE broadcasting. Building my own firmware with changed flags is OK for me, but I didn't see any `MICROPY_PY_ENABLE_UBLUETOOTH` flag in the `mpconfig.h`/`mkconfigboard.h`/`mpconfigport.h` files.

So, I want to have an option to enable `ubluetooth`. Is this hard to do?

laurensvalk, 2025-06-11:
> _Originally posted by @laurensvalk in [#140](https://github.com/pybricks/support/issues/140#issuecomment-889827027)_
>
> Why you won't be enabling `ubluetooth`? It is hard to enable it?

We want to make sure users can't break the connection that's needed to download and run programs. 

We aim for a smooth coding experience where resetting or removing the batteries isn't needed half the time. If you've been coding with other micropython boards, you might know what I mean. :smile:

> Because I want to transfer lots of data not possible with Pybricks BLE broadcasting.

You could use this method to transfer more data: https://pybricks.com/projects/tutorials/wireless/hub-to-device/pc-communication/. This should even be a bit easier than building your own Bluetooth handlers all in micropython.

laurensvalk, 2025-06-12:
Why not? 

You can still combine that with broadcasting if you like, or have the pc connect to both hubs.

Enabling ubluetooth isn't possible.

romw314, 2025-06-13:
> Why not?
> 
> You can still combine that with broadcasting if you like, or have the pc connect to both hubs.

I mean not possible with communication between 2 hubs. PC is not an option as I want the project to be started as simple as turning on all hubs and starting programs, or connecting power supply, but I can try to use a Raspberry Pi as a bridge.

> Enabling ubluetooth isn't possible.

Ok.