pybricks/support #2252: [Feature] Simplifying bluetooth tasks

laurensvalk, 2025-06-30:
We have a task queuing system in `pbio/task` that we ended up using only for Bluetooth. In porting this to the new API in #2251 (which we _could_ do verbatim), I started thinking about potentially simplifying this, reducing code size, and reducing the risks of problems when programs end. We might not do this, but I wanted to collect some thoughts before I forget about this.

We don't allow users to queue different Bluetooth tasks, so in principle the only things we'd need to queue are status messages, stdout and up to _one_ task such as "scan and connect", or "start broadcasting". Instead of queuing multiple of the latter, we could have the caller await the one task become available.

Ideally, this would reduce the tasks management (now a mix of `pbdrv/bluetooth____`, `pbsys/bluetooth`, `pbio/task`) to just one level, similar to:

```c
// bluetooth_process:

for (;;) {

    if status was set or timer expired
       await send status 

    if stdout was buffered
       await send stdout

    if current_task:
       await current_task
       clear current_task
}

pbio_error_t start_task(task) {
   if (current_task) {
       // busy, try again later
       return PBIO_ERROR_AGAIN
   }
   current_task = task
}

```

With at most one "task" (i.e. anything other than status or stdout) active, we shouldn't have issues such as having to make static task singletons for things like the remote, or use after free issues. We would also only ever have to cancel the current task, not cancel pending tasks (there are some open todos for this).

dlech, 2025-06-30:
I would be a bit worried that doing it this way could lead to resource contention where one task hogs the current_task and any other tasks never run or are delayed unnecessarily. I.e, if there is user code in a tight loop that starts a new task as soon as the previous one finishes, any tasks in other coroutines would never have a chance to run.

laurensvalk, 2025-06-30:
Starting another Bluetooth task from another user co-routine currently raises an exception ("can use this resource only once"), so it doesn't seem like we would lose functionality this away. We have it that way currently as an extra precaution since otherwise we could have many Bluetooth methods trying to access the same singleton and bad things might happen.

We could keep it that way so this issue never happen silently. Or we could allow it and document it, since in principle it could now be done safely, and work fine if you added a wait. For example, we had a question at one point about broadcasting and setting the remote light. At the moment, this requires extra work from the user to only ever do this from one task.

laurensvalk, 2025-07-03:
Expanding and keeping #2255 in mind, it could be done like this at the `pbdrv` level.

```
for (;;) {
  
    reset bluetooth

    wait until advertising requested

    start advertising

    wait for connection or request for stop advertising

    stop advertising

    while connected {

        if status was set or timer expired
           await send status 

        if stdout was buffered
           await send stdout

        if current_task:
           await current_task
}
```

Then it is independent of the program-run logic. Currently program logic is mixed in with `pbsys/bluetooth` (which could go away).

This also takes the `reset bluetooth` logic into account. At the moment, this is done by having  `pbsys/bluetooth` call down to `pbdrv/bluetooth_` by exiting the Bluetooth SPI process, which has this ominous note:

```
        // REVISIT: should probably gracefully shutdown in case we are in the
        // middle of something
        process_exit(&pbdrv_bluetooth_spi_process);
```

This also gets a bit closer to the USB logic, which already has the status sending included (but we can still clean up the _order_ of abstractions, so `pbsys` would tell `pbdrv` to send a status, so not have `pbdrv` monitor `sys` status changes.)