pybricks/support #2259: [EV3 - help wanted]: Enable PRU1 for LED PWM

laurensvalk, 2025-07-05:
The EV3 has two PRUs. The LEGO firmware, ev3dev, and Pybricks use PRU0 to provide UART on sensor ports S3 and S4. We use the same ready-made binary in Pybricks for UART0, but with a lot less less wrapper code since we don't have Linux. Some improvements _could_ be made there (let me know if you're interested) but this is working well enough so we can leave it for now.

[ev3dev uses PRU1](https://github.com/ev3dev/ev3dev-pru-firmware/blob/master/ev3-pru1/main.c) to provide PWM brightness for the button LEDs and to keep track of motor position.

We don't necessarily need the motor positions (we use GPIO IRQs now, which works well enough), but it would be nice to mimic the LED brightness control so we can [complete the LED PWM driver for EV3](https://github.com/pybricks/pybricks-micropython/blob/master/lib/pbio/drv/pwm/pwm_ev3.c).

From its readme:

> This requires [TI CSS6](http://processors.wiki.ti.com/index.php/Download_CCS) with PRU Compiler Tools 2.1.2 to build. (Use the web installer to install the minimum required components.)

It doesn't seem to build with more recent versions anymore.

So here are some pointers to tasks to getting this to work.

- Are there any other OSS command line tools that we could use to build this?
- Get [ev3dev PRU1](https://github.com/ev3dev/ev3dev-pru-firmware/blob/master/ev3-pru1/main.c) code to  build.
- Remove the `pru_rpmsg` waits and kick IRQs. I think we can just keep the loop that reads from shared memory to keep it simple. (Thoughts?). We also don't need to tell Linux where the address is. We can just use the same address on host and PRU.
- Build it and include the binary with the firmware
- Load it to the appropriate address and start the PRU
- See also the [`am18x-lib` ](https://github.com/pybricks/am18x-lib-ev3) as an alternative to TI StarterWare which has some minimal code for loading PRU binaries.
- Ensure we don't use the shared memory regions in other code.

In previous hacks, I have tried to take the ready-made `.fw` file from the ev3dev kernel, which looks like an ELF file, bundled it with the firmware, and copied it to the PRU peripheral. I got the LEDs to give some sign of life, but it was incomplete and very hacky. It would be nice to do it properly.

Nobody is currently working on this, so feel free to share your interest and we can assign you to this issue. Many thanks!

ArcaneNibble, 2025-07-05:
> Are there any other OSS command line tools that we could use to build this?

There is a [GCC port for the PRU](https://github.com/dinuxbg/gnupru), but I don't know if it's been used or tested with the AM1808. It seems to be mostly getting used and tested with the AM335x on the BeagleBone.

dlech, 2025-07-13:
Related issue: https://github.com/pybricks/support/issues/67

In Pybricks v2.x, we consider the EV3 to have "one" LED even though the left and right side can be independently controlled. Do we want to stick to that decision? FWIW, we made a similar decision on SPIKE Prime where the main button technically has 2 LEDs but they are controlled as one.

I'm inclined to keep it simple so `hub.light.on()` works like other hubs where there is only one light (minus the blue component of course). But I would not be opposed to adding some kind of back door for people to have a little fun with it and control the lights independently.

laurensvalk, 2025-07-13:
> I'm inclined to keep it simple so `hub.light.on()` works like other hubs where there is only one light (minus the blue component of course).

Yes, this is precisely the aim. This is how the current draft implementation works too, except there isn't PWM yet.

> But I would not be opposed to adding some kind of back door for people to have a little fun with it and control the lights independently.

`hub.light.left` and `hub.light.right` could be attributes that are themselves color lights. Could be enabled on SPIKE as well.

dlech, 2025-07-14:
IIRC, the PRUs only have 2k instruction RAM, so any kind of setup/initialization we can do outside of the PRU is beneficial. So yeah, coordinate.

ArcaneNibble, 2025-07-14:
Right, but do we need to specifically share the "34" half of Timer64 number 0? This would imply that the timer needs to run at a rate acceptable for both the systick *and* the PWM.

Or can we change it to use (for example) the "12" half, or to use one of the other timers?

laurensvalk, 2025-07-14:
> Right, but do we need to specifically share the "34" half of Timer64 number 0? 

No, they can be separate. We could use 12 or another timer to provide the time for `pbdrv/clock` if that lets us keep the PRU1 code unchanged.

ArcaneNibble, 2025-07-14:
> It would be nice if we could just use the existing PRU firmware without having to rebuild that.

Is the goal to reuse existing PRU1 binaries, or should we attempt to simplify the code? imo it would be really nice to have binaries that can not only be built using f/oss tooling but also an interface that doesn't involve the rpmsg infrastructure (as suggested in the initial issue)

laurensvalk, 2025-07-14:
I'm leaning towards rebuilding if we can use f/oss tools for it.

But if we can't:

> an interface that doesn't involve the rpmsg infrastructure

then this isn't the end of the world either. If I'm reading it right, all we'd have to do is set one flag in shared memory to push it past this point:

```c
	status = &resource_table.rpmsg_vdev.status;
	while (!(*status & VIRTIO_CONFIG_S_DRIVER_OK));
```

and then it just stays in the loop that updates the lights. This is what my earlier hacked attempts relied on.

We shouldn't have to implement the kick (and response) for it to tell us where the shared memory address lives since we know that already.

ArcaneNibble, 2025-07-14:
LED blinky hello world from PRU1, *using the GCC toolchain* (no TI proprietary compiler needed)

https://github.com/user-attachments/assets/4d764166-45f3-4ee7-be97-c78fd098ef20

ArcaneNibble, 2025-07-14:
Given that it clearly *is* possible to build some new PRU firmware, would it make sense to use the PRU to offload tasks such as feeding the audio PWM in the future?

ArcaneNibble, 2025-07-14:
> Did you have to build your own toolchain for this?

Yes, I've duct-taped together a custom toolchain build using the shell scripts [here](https://github.com/dinuxbg/gnupru) because I am running on an Apple Silicon Mac. However, this project has x86_64 Linux prebuilt binaries that we should be able to use.

laurensvalk, 2025-07-14:
Awesome. Will have to look at the details later.

For what it's worth, I am not against committing this binary to git as long as we can document the steps to build it if we have to. Especially if it is just LEDs we're going for, which should be fairly final at some point. (But I saw the note about audio 😄)

dlech, 2025-07-14:
There probably won't be enough room left, but it would be cool if we could cram I2C in there as well.

> If we go this way, the blocker is going to be PRU toolchain devops. Let me know if you want me to take a look at that.

Go for it. I suggest to manage the PRU source code separately at https://github.com/pybricks/pybricks-pru and set up a publish script like I did [for u-boot](https://github.com/pybricks/u-boot/blob/pybricks-ev3/.github/workflows/publish.yml).

ArcaneNibble, 2025-07-14:
> I2C

Is this for NXT backwards-compatibility? Do you have any notes/references about how "normal" vs "borked" the typical Lego I2C device is? (For example, more about "A secondary reason for this is that the NXT Ultrasonic sensors does not work well with standard I2C masters. It requires an extra clock pulse between the stop and the next start. (TODO: add references here.)" from the ev3dev kernel handbook?)

dlech, 2025-07-15:
This is the I2C implementation from ev3dev: https://github.com/ev3dev/ev3-kernel/blob/8f4dfb355aae339ebff6543ed3edfaf0847a95c9/arch/arm/mach-davinci/legoev3-fiq.c#L257

It includes the extra clock pulse needed for the NXT ultrasonic sensor. I'm not sure there is much more documentation on that outside of other EV3 and NXT firmware source code.

laurensvalk, 2025-07-15:
> Is this for NXT backwards-compatibility?

Yes, but not just that. Most third-party devices for EV3 use I2C (Mindsensors, HiTechnic). Some of those are generic I2C devices without LEGO quirks. We expect to expose the generic interface with `pybricks.iodevices.I2CDevice`. And it will be used when some devices are auto-detected, like the official LEGO temperature sensor.

(NXT sensor ports are fairly similar to those on EV3. Only they have hardware I2C instead of hardware UART. But since the EV3 UART sensors can also run at 2400 baud, I'd be curious to see if we can make a soft UART work well enough on NXT. That way, "if the cable fits, it just works". If we want to experiment with this, I suppose we could try out a GPIO / timer IRQ implementation on EV3 first to see how well this would work.)

ArcaneNibble, 2025-07-18:
This has been merged and can be closed now

dlech, 2025-07-18:
Yup, and let's start a new issue to discuss if we want to add `hub.light.{left,right}` to be able to control them separately.

closed 2025-07-18 by dlech

ArcaneNibble, 2025-07-24:
That's weird, because it's behaving as expected when I test it (`hub.light.on(Color.YELLOW*0.2)` turns both LEDs on to a faint yellow)

BertLindeman, 2025-07-24:
What I see on [git hash 7c11ba20](https://github.com/pybricks/pybricks-micropython/actions/runs/16498102301) with:

```python
 for x in range(10):
     hub.light.on(Color.YELLOW*(1/(x+1)))
     print(x)
     wait(1000)
``` 
Two similar colors.
Starting really orange and getting more to green at the lower values.