pybricks/support #2260: [EV3] Sort out caches, cache coherency with DMA, memory barriers, atomicity, and other memory model gremlins

ArcaneNibble, 2025-07-05:
Memory caches are very important for getting good performance from modern CPUs, but they also introduce additional complexity because caches introduce opportunities for different agents to have a different view of what memory contains. Additionally, when there are multiple agents all accessing memory, complexity can arise if they try to access the same memory locations at the same time. Although the ARM926EJ-S is a relatively simple CPU in a single-core configuration, the AM1808 SoC contains multiple DMA controllers and other bus mastering peripherals. These concerns will be of increasing relevance as we make additional use of them.

In order to prevent extremely-difficult-to-debug bugs in this area, we need to have a clear plan for how this will be approached, such as by doing the following:

- [ ] Figure out what state the caches are currently being left in, and enable them if they are not enabled.
- [ ] Check if DMA controllers are cache-coherent with the CPU, or if CPU caches need to be flushed before performing DMA
- [ ] Write an internal API for managing memory issues around DMA transfers. When writing data to be accessed by peripherals, this requires at minimum a _compiler_ memory barrier and possibly also a cache flush. When reading memory written by peripherals, this may require a cache invalidate or an uncached mapping.
- [ ] Figure out what happens if two agents (e.g. the CPU and a DMA controller) access the same memory at the same time (which is currently used by the ADC implementation). Are word accesses atomic, or can torn values be observed?
- [ ] Decide what memory model will be followed, in order to make sure the compiler does not miscompile code (for example: adopting the C/C++ memory model which requires atomic types, adopting the Linux kernel memory model, adopting a simplified memory model tailored to GCC/clang behavior)

dlech, 2025-07-06:
So I made https://github.com/pybricks/pybricks-micropython/commit/1d1065461cafe68d68db5dda4e3615c0a031a126 to fix a regression caused by switching from using the Linux loader in U-Boot to using a generic ELF loader. I'm not sure if all of it was necessary, but clearly we have some work to do before we can enable caches. But at least now, we know for sure where we are starting from.

laurensvalk, 2025-07-07:
See also this inspiration for our DMA display driver which [handles the cache each time prior to writing to the display](https://github.com/pybricks/ev3rt-lib/blob/72d28a64e521490acb586e9636e836b92e1384cc/target/ev3_gcc/drivers/lcd/src/lcd_spi.c#L177).

dlech, 2025-08-05:
Does it make sense to revert https://github.com/pybricks/pybricks-micropython/commit/1d1065461cafe68d68db5dda4e3615c0a031a126 now?

ArcaneNibble, 2025-08-05:
We still need to turn the MMU *off* before we turn it back on, so I figured we should leave it?

dlech, 2025-08-05:
Fine with me to leave it if there isn't a better place for it. We should probably update the comment though to reflect that though.

ArcaneNibble, 2025-08-05:
https://github.com/pybricks/pybricks-micropython/pull/369