pybricks/support #2265: [Feature] Implement awaitable ADC reading

laurensvalk, 2025-07-08:
The method `pbdrv_adc_update_soon` is a bit of a hack to get a value sooner than the typical `10 ms` sample time. It mainly exists to support the NXT color sensor (on EV3). It does 4 sequential ADC measurements to measure light intensity while the LED is red/green/blue/none, and we would rather not have it take 40 ms per color measurement. The current implementation in `lib/pbio/src/port_dcm_ev3.c` is this:

```c

pbio_error_t pbio_port_dcm_await_new_nxt_analog_sample(pbio_port_dcm_t *dcm, pbio_os_timer_t *timer, const pbdrv_ioport_pins_t *pins, uint32_t *value) {
    pbio_os_state_t *state = &dcm->child;

    PBIO_OS_ASYNC_BEGIN(state);

    // Wait for LED to settle.
    PBIO_OS_AWAIT_MS(state, timer, 1);

    // Request a new ADC sample. Revisit: Call back on completion instead of time.
    pbdrv_adc_update_soon();
    PBIO_OS_AWAIT_MS(state, timer, 4);

    // Get the value.
    uint8_t pin = dcm->category == DCM_CATEGORY_NXT_COLOR ? 6 : 1;
    *value = pbio_port_dcm_get_mv(pins, pin);  // this calls pbdrv_adc_get_ch and just scales it

    PBIO_OS_ASYNC_END(PBIO_SUCCESS);
}
```

Instead of doing it that way, perhaps we could have a complimentary function to `pbdrv_adc_get_ch` called something like `pbdrv_adc_await_ch` that will handle this at the `pbdrv/adc_ev3` level and return the value when it is ready. Any ADC settling time (however long that is?) could be handled there as well, so the application code can just await it instead of hard coding this `4 ms` wait.

Other ideas for doing this better would be welcome too.

Side note: would it make sense to scale ADC measurements to mV at the driver level instead of having the application level do it?

laurensvalk, 2025-07-08:
Increasing poll frequency sounds good.  Then we can drop the `poll_soon` altogether.

> run all of the pbio_port_dcm_await_new_nxt_analog_sample() protothreads in parallel

This is more or less what's happening at the moment. It runs as part of that sensor port's process and all ports run in parallel.

dlech, 2025-07-08:
> This is more or less what's happening at the moment.

Very good. I must not be very awake yet and didn't notice there wasn't in index variable in the for loop. 😪

laurensvalk, 2025-07-11:
Completed via https://github.com/pybricks/pybricks-micropython/pull/322

laurensvalk, 2026-09-21:
This is done, and used for NXT Color and Light sensor.

closed 2026-09-21 by laurensvalk