pybricks/support #2279: [Doc] Collect "hardware hacker" documentation on Lego I/O and peripherals

ArcaneNibble, 2025-07-16:
I've been trying to understand:

* how many "generations" of Lego electronic peripherals exist, and which ones are compatible
* how the I/O ports on the EV3 work
* how peripheral auto-detection works

but I've been finding the relevant information scattered and somewhat unclear. I hope that this issue can be used to collect information to the detail level that would be useful for a hardware hacker.

# Hardware generations

## RCX-era

These devices connect using two conductors arranged into a 2x2 stud square connector with conductors along the sides of the studs.

These parts seem to use the "legacy" (bright solid colors) design language.

This generation seems to also be called the "9V system"?

Motors work by applying power across the two conductors.

Sensors can be "unpowered" or "powered". Sensors report analog values. TODO: Electrical characteristics of the inputs?

## Power Functions

These devices use a connector that has the size of a 2x2 stud square, but the connector only has two plastic studs. The other half of the connector is replaced with an oval shape with four pins along the inside of the oval.

These parts seem to use the "NXT-era" (orange as the accent color) design language.

Motors appear to use the two additional pins to control the direction.

TODO: Are there sensors that are part of this system?

TODO: How does the adapter cable (Lego part 8886) work? How is it wired?

## NXT

These devices use a modified RJ11 with an offset tab. Cables have 6 wires.

These parts seem to use the "NXT-era" (orange as the accent color) design language.

Motors now contain encoders which send back digital pulses as they rotate.

Sensors can be either analog or digital using I2C. The NXT HDK mentions the ability to support RCX-style active sensors using an adapter cable.

TODO: Is the adapter cable "x1676" or "54690" usable with both motors and sensors? How is it wired?

## EV3

These devices use the same modified RJ11 as the NXT.

These parts seem to use the "EV3-era" (red as the accent color) design language.

Motors appear to be interchangeable with the NXT.

Digital sensors now use UART communications, but the pinout is compatible with NXT sensors.

TODO: Where do you find documentation on the UART protocol?

TODO: Why does ev3dev mention a difference between "EV3/Analog" sensors and "NXT/Analog" sensors?

TODO: Why doesn't the RCX converter cable (is this the "x1676"/"54690" cable??) work with the EV3 according to the [ev3dev docs](https://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensors.html#rcx-analog-sensors)? What is the incompatibility?

## Powered Up

These devices use a custom 6-pin connector.

This seems to also be called "Lego Power Functions 2.0"?

These parts seem to use a "gray and white" design language in the "mainstream" product lines, teal accents in the "home education" Mindstorms product line, and yellow accents in the "classroom education" Lego Education product lines?

TODO: Are parts fully compatible across the product lines? (when using Pybricks / when using the official software)

TODO: Is the same pinout used for motors and sensors now?

TODO: How do motors with encoders/feedback work in this system?

TODO: This pinout seems like a hybrid of the EV3 sensor and motor pinouts. Do adapters exist? How compatible/different are the EV3/Powered-Up protocols?

## WeDo

TODO: What is WeDo??

# EV3 I/O port circuitry

TODO

ArcaneNibble, 2025-07-16:
One thing I'm notably confused by in the EV3 schematics is why the EV3 sensor circuitry has *both* a path directly connecting pins to the port (`DIGIxn`) as well as the 74HC125 buffers (`TXINx`, `RXINx`)

dlech, 2025-07-16:
> Lego IR protocols

Philo's page has a copy of the Power Functions IR protocol docs from LEGO. The EV3 remote/IR Sensor is partially compatible with that.

https://philohome.com/pf/LEGO_Power_Functions_RC.pdf

laurensvalk, 2025-07-17:
> IIRC, the BOOST Color/Distance sensor can transmit some compatible commands but not receive them.

We support this 😄 

[image]

laurensvalk, 2025-07-17:
@ArcaneNibble, writing it all down could probably be whole book or two, but feel free to pick our brains on specifics as you run into them. Either here or on a video chat.

Our understanding of the bigger picture and a reasonable way to deal with LEGO sensors in code has gradually evolved and changed a couple of times over the years, but it is starting to get in a good place now. I'm happy to share the high level ideas as well as implementation specifics.

It is usually a balance of making things work....
- ... simple and effective for beginning users (e.g. useful classes available out of the box)
- ... configurable and extensive for advances users (e.g. calibrate your own colors if you want)
- ... consistent across platforms and generations (e.g. color sensors with 4 different protocols)
- ... while dealing with the quirks of each sensor (e.g. data split across different device modes, protocol bugs...)

The motor controllers are our own (which is a bit of a saga in itself ☕ ). We generate a basic trajectory and follow it using a PID controller with a feedforward component based on a simple DC motor model with a state observer. Other than measuring model parameters and configuring control constants for the EV3/NXT motors, this code is in good shape and already enabled on NXT/EV3.

> TODO: Why does ev3dev mention a difference between "EV3/Analog" sensors and "NXT/Analog" sensors?

In the end, there was only ever one EV3 analog sensor, the Touch sensor. Its value is measured on pin 6. The NXT and all third party sensors used the ADC value on pin 1. See [code](https://github.com/pybricks/pybricks-micropython/blob/448455274efc5ec35d9e394cec3bf77b548dde14/lib/pbio/src/port_dcm_ev3.c#L522).

> TODO: Are parts fully compatible across the product lines? (when using Pybricks / when using the official software)

With the official products, each theme or even set has its own app. Almost nothing is compatible, even if it has the same plug. Believe it or not, if the plastic has the wrong color you're out of luck on a $400 set (the Inventor Hub is the same as SPIKE Prime save for a USB device ID so you can't use it with the SPIKE app). Since the past 5 years or so, it is app-store only (no installers), so things just die when the app is pulled. This has already happened for a few sets (Boost Star Wars).

Pybricks' raison d'être 😄 

Our motto is **if the cable fits, it should just work**. We're pretty much there for Powered Up (except for WeDo 2.0 which can't be updated). For the NXT/EV3 generation we might get very close too. The only open question might be soft-UART support for NXT.

dlech, 2025-07-18:
### RCX

> This generation seems to also be called the "9V system"?

This is more commonly used among train enthusiasts rather than RCX, I think, but same connector, so same system.

> TODO: Electrical characteristics of the inputs?

The NXT Hardware Developer SDK has some info on how these work. Passive sensors are basically just measuring resistance of the sensor and active sensors are powered for 3 ms, then read for 0.1 ms.

### Power functions

> Motors appear to use the two additional pins to control the direction.

And on the Power Functions servo motor, they control the position. And they serve as data lines for the WeDo sensors.

> Sensors using this connector existed as part of the "WeDo 1.0" system. TODO: How do they work?

I think we mostly figured this out through reverse engineering. There is a WeDo USB hub that can connect to senses and drive small motors (no separate power, so 5V 500mA!). We wrote [ev3dev drivers](https://github.com/ev3dev/lego-linux-drivers/tree/ev3dev-buster/wedo) for it that should fill in some of how it works.

> TODO: How does the adapter cable (Lego part 8886) work? How is it wired?

It just connects the two power wires and not the control wires.

### NXT

> Sensors can be either analog or digital using I2C.

The NXT Color sensor is a bit of a hybrid analog sensor where you can also fetch the calibration table and control which LED is turned on. It uses it's own bit-banging protocol rather than I2C.

### EV3

> TODO: Where do you find documentation on the UART protocol?

We made our own: https://github.com/pybricks/technical-info/blob/master/uart-protocol.md

The EV3 source code kind of documents it, but the code comments there seem to be for an earlier design revision and didn't get update to how it was actually implemented.

### Powered Up

> This seems to also be called "Lego Power Functions 2.0"?

This was a pre-release name before they went all-in on the Powered Up branding.

> TODO: Are parts fully compatible across the product lines? (when using Pybricks / when using the official software)

Pybricks says, if it fits the plug, it works! LEGO, not so much. Official LEGO apps tend to only support the accessories that shipped with the kit that the app is targeted to and not the full ecosystem.

> TODO: EV3 <-> Powered Up adapters

From a reputable vendor: https://www.mindsensors.com/home/28-spike-prime-sensor-adapter-for-ev3.html

ArcaneNibble, 2025-07-18:
> It just connects the two power wires and not the control wires.

Do Power Functions motors just spin in one direction if the control lines are floating?

Was there never an autonomous "intelligent" hub that used these connectors? Or were there only train/model "battery boxes", remote controls, and the WeDo 1.0 hub?

dlech, 2025-07-18:
> Do Power Functions motors just spin in one direction if the control lines are floating?

I think the motor would just not run. Only the C1/C2 lines are connected to the motor. See the diagram at https://www.philohome.com/pf/pf.htm

The other 2 wires power the infrared receiver (and the servo motor).

> Was there never an autonomous "intelligent" hub that used these connectors? Or were there only train/model "battery boxes", remote controls, and the WeDo 1.0 hub?

Correct, no 1st-party programmable brick. But there are some 3rd-party ones, like Buwizz.

ArcaneNibble, 2025-07-19:
Ohhhh, PF motor != PF servo

This now all makes sense electrically, thanks!

ArcaneNibble, 2025-07-28:
Could someone look over the edits I've made to the original post to see if there are any errors?

BertLindeman, 2025-07-28:
The Power Function connector I might call it a plate, but it is a bit thicker, so forget it. But an important thing about this connector is that it can be stacked to connect more than one device.
Like this:

[image]

Not too important, but it **was** at the time.

dlech, 2025-07-29:
One difference I noticed between EV3 motors and NXT motors is that the EV3 motors also have a Schmitt trigger in the motor itself on pins 5/6. This makes using the EV3 motors with custom electronics easier, e.g. you can connect it directly to a quadrature encoder without having to add your own Schmitt trigger.

dlech, 2025-07-29:
> Pin 1 is the pin normally used for passive analog sensors. It is also capable of supplying power to RCX active sensors, which can be connected using an adapter cable. Unlike the RCX, the 10k pull-up resistor cannot be disabled.

Some I2C sensors, like the LEGO NXT Ultrasonic sensor, also require extra power on pin 1.

dlech, 2025-07-29:
> The NXT HDK mentions that pin 6 is connected to an analog-capable pin on the NXT CPU, but this does not appear to ever have been enabled in software.

Until Pybricks, that is. 😄

dlech, 2025-07-29:
> Powered Up

There are also some non-UART Powered up devices. The WeDo 2.0 Medium motor (aka Simple Medium Linear Motor - 45303)  and the Powered Up Train Motor (88011) do not have speed feedback. And of course the Powered Up Light (88055) doesn't use UART. 

There are a few other unused device IDs that we got names for by reverse-engineering the official LEGO firmware, but no products have a appeared that use any of these since then.

```c
static const lego_device_type_id_t legodev_pup_type_id_lookup[3][3] = {
    /* ID1 */ [DEV_ID_GROUP_GND] = {
        /* ID2 */ [DEV_ID_GROUP_GND] = LEGO_DEVICE_TYPE_ID_LPF2_POWER,
        /* ID2 */ [DEV_ID_GROUP_VCC] = LEGO_DEVICE_TYPE_ID_LPF2_TURN,
        /* ID2 */ [DEV_ID_GROUP_PULL_DOWN] = LEGO_DEVICE_TYPE_ID_LPF2_LIGHT2,
    },
    /* ID1 */ [DEV_ID_GROUP_VCC] = {
        /* ID2 */ [DEV_ID_GROUP_GND] = LEGO_DEVICE_TYPE_ID_LPF2_TRAIN,
        /* ID2 */ [DEV_ID_GROUP_VCC] = LEGO_DEVICE_TYPE_ID_LPF2_LMOTOR,
        /* ID2 */ [DEV_ID_GROUP_PULL_DOWN] = LEGO_DEVICE_TYPE_ID_LPF2_LIGHT1,
    },
    /* ID1 */ [DEV_ID_GROUP_PULL_DOWN] = {
        /* ID2 */ [DEV_ID_GROUP_GND] = LEGO_DEVICE_TYPE_ID_LPF2_MMOTOR,
        /* ID2 */ [DEV_ID_GROUP_VCC] = LEGO_DEVICE_TYPE_ID_LPF2_XMOTOR,
        /* ID2 */ [DEV_ID_GROUP_PULL_DOWN] = LEGO_DEVICE_TYPE_ID_LPF2_LIGHT,
    },
};
```

Speaking of IDs on Powered Up, there are lots of posts around the internet that imply there is an "ID resistor" as if the port detection was somehow measuring the resistance value of such a resistor. This likely comes from the assumption that the Powered Up port ID works like EV3. But this is simply not true. Powered Up ports don't have any analog inputs and the port ID algorithm uses purely digital I/O.

ArcaneNibble, 2025-07-29:
> Some I2C sensors, like the LEGO NXT Ultrasonic sensor, also require extra power on pin 1.

How do you know if you need to turn the extra power on? Is it hardcoded according to the type of the device?

dlech, 2025-07-29:
> Is it hardcoded according to the type of the device?

For NXT I2C sensors, it is just hard-coded. There isn't any kind of device info we can get from the sensor other than the name and manufacturer. 

There are Powered Up UART devices that have similar requirements (but not any EV3 UART devices that I recall) that also need power on pin 1 or pin 2. They have device info that is sent from the sensor that includes a bit flag to indicate this.