pybricks/support #2280: [Feature] EV3 USB Host Port

laurensvalk, 2025-07-17:
This might be day dreaming for the moment, but while our minds are on USB, there is the host port to consider.

** USB HID Devices **

If there was a `pybricks.iodevices.HIDDevice` class then users could use keyboard or controller input.

** Hub to hub communication **

The host port of one brick could connect to the client port of another. Or use a USB hub if you want to really go crazy. If we made something that worked well with the rest of our Pybricks USB protocol, then the client could also be a SPIKE or NXT brick.

** Hub to hub communication (classic daisy-chaining)**

The original labview-based software had a way of remote-controlling the motors and sensors of downstream daisy-chained hubs. So hub B below could see and control the devices on C and D. You had to boot them up in the right order (D first) and it was very slow. The newer EV3 scratch app dropped support for this.

```
A ---> B ---> C ---> D
```
I would really rather not do it this way, but I wanted to mention it here for historical context.

dlech, 2025-07-17:
https://docs.tinyusb.org/en/latest/index.html seem pretty popular.

ArcaneNibble, 2025-07-18:
Dreaming far into the future, what about USB Mass Storage support? This will *definitely* require a plan for [filesystems and mount points](https://github.com/pybricks/support/issues/2261)

laurensvalk, 2025-07-18:
> Do you think we can find an off-the-shelf USB host stack for microcontrollers? 

There's always your favorite TI library which includes host examples 😆 

> Dreaming far into the future

USB hubs, with one brick to control them all... If the host implements the Pybricks protocol, it could send new programs to the client bricks. Very convenient, and also 12 years late.