pybricks/support #2295: [Bug] pybricksdev waits ~30 seconds after flashing

BertLindeman, 2025-07-25:
**Describe the bug**
Flashing recent firmware, pybricksdev waits on something after "restart ev3.... Done" it takes 30 seconds or a bit more.
After that time pybricksdev ends normally.

**To reproduce**
Steps to reproduce the behavior:
1. On linux mint with pybricksdev 1.2.0 installed
2. in a terminal enter ` pybricksdev flash ev3-firmware-build-4054-gitaecbe204.zip` or another firware zip
3. See the firmware being installed up to `Restarting EV3...Done.`
4. Wait for pybricksdev to end >30 seconds.

**Expected behavior**
pybricksdev to end soon after `Restarting EV3...Done.`

**Screenshots**
There is a saying that a picture is worth a 1000 words. Screenshots really help to identify and solve problems.

**Extra info**
Older firmware does not have the long wait, like
`ev3-firmware-build-3989-git44845527.zip`.
Maybe related to the latest changes for EV3?

closed 2025-07-26 by dlech

dlech, 2025-07-26:
> After flash of NXT

NXT and EV3 have completely different USB controllers, so all bugs with NXT USB should be reported separate from EV3 USB.

BertLindeman, 2025-07-26:
> > After flash of NXT
> 
> NXT and EV3 have completely different USB controllers, so all bugs with NXT USB should be reported separate from EV3 USB.

Will report the NXT separately if it still occurs after the latest USB update.