pybricks/support #2306: [Bug] EV3 USB program starts before first status received

laurensvalk, 2025-08-04:
On EV3, we are **not** getting a status before the program starts, so we start slot 0 by default.

```
pybricksdev run usb /home/laurens/git-pybricks/pybricksdev/demo/longdemo.py 
100%|████████████████████████████████████| 206/206 [00:00<00:00, 117kB/s]
starting:  None
using selected 0
STATUS array('B', [0, 64, 0, 0, 0, 0, 2]) # <----- selected slot is 2, but this is too late
```

On SPIKE, we are getting a status event before we download the program, so it is starting slot 2:

```
pybricksdev run usb /home/laurens/git-pybricks/pybricksdev/demo/longdemo.py 
STATUS array('B', [0, 8, 0, 0, 0, 0, 2])
running 0
selected 2
100%|████████████████████████████████████| 206/206 [00:00<00:00, 42.8kB/s]
starting:  None
using selected 2
```

laurensvalk, 2025-08-04:
Neither the Bluetooth driver nor the USB driver should be managing this, but yes, it is supposed to send when [the status changes](https://github.com/pybricks/pybricks-micropython/blob/ccafa60a1ccbd71f2f04df5f29d4ac11f9f3dd24/lib/pbio/sys/bluetooth.c#L198-L203). This is why we're working on pulling this out and `sys/bluetooth` should go away. At the moment, I think `pbdrv/usb` has (or should for now) a copy of this logic.

Instead, when a status is set, `pbsys/host` will call both `pbdrv_usb_schedule_status_update` and `pbdrv_bluetooth_schedule_status_update`. The respective driver can then send it when necessary. For example when it is different from the previously transferred message or repeat it after some timeout. The bluetooth driver [might do it this way](https://github.com/pybricks/support/issues/2252#issuecomment-3032334647).

ArcaneNibble, 2025-08-04:
Ah, the USB logic doesn't check for slot change, only status flag change.

With that being said though, it *does* send a status message when the application subscribes. The fact that pybricksdev isn't picking that up is rather strange.

closed 2025-08-07 by dlech