pybricks/support #2308: [Bug] Figure out MicroPython module alignment requirements

ArcaneNibble, 2025-08-05:
During the investigation of a crash occurring on EV3, we found that there are potential issues with the data alignment (i.e. to a multiple of 4 bytes) of MicroPython modules. See [here](https://github.com/pybricks/support/issues/2301#issuecomment-3156265178).

This may affect modules containing native code.

This needs to be properly specified, and any potential breaking changes need to be coordinated between the firmware and `pybricksdev`.

ArcaneNibble, 2025-08-05:
Not only do we need to check if the overall size is aligned, but we may also need to check if the data which comes after the module name (i.e. the result of `mpy_data_get_buf`) needs to be aligned.

laurensvalk, 2025-11-20:
> Not only do we need to check if the overall size is aligned, but we may also need to check if the data which comes after the module name (i.e. the result of `mpy_data_get_buf`) needs to be aligned.

It looks like the `mpy_reader` reads it [byte-by-byte](https://github.com/pybricks/micropython/blob/13580b6ad057173f62e8b2363e01d6851bcc6699/py/reader.c#L42-L68). So this part of the problem should be OK.

However, if we're revising this as part of #2364 we can still fix alignment once and for all.