pybricks/support #2342: Implement hub name support for EV3,

dlech, 2025-08-30:
To allow users to give their EV3 a unique name, we need to implement the hub name feature in the firmware like we do in other hubs.

This involves creating a special section in the firmware that contains the default hub name. Then the script that creates the firmware metadata needs to be updated to include the `hub-name-size` and `hub-name-offset` read from the linker script or some other build output.

jaguilar, 2025-11-24:
Could this be implemented as a new field in [_pbsys_storage_settings_t](https://github.com/pybricks/pybricks-micropython/blob/19dfe5477a290edc98091f136cc440067cb6e4d8/lib/pbio/include/pbsys/storage_settings.h#L37)? It seems potentially simpler than linker script shenanigans.

laurensvalk, 2026-09-21:
It has a name now! So we can close this.

closed 2026-09-21 by laurensvalk