pybricks/support #2364: [Bug] Multi mpy files always start with __main__

laurensvalk, 2025-09-29:
**Describe the bug**
The first (or only) module in a program is always called `__main__` rather than the actual script name.

This is not to be confused with `__name__ = "__main__"`, which is an entirely separate mechanism. This does not reference at our MPY6 metadata name at all.

**Expected behavior**
We should use the actual name when compiling in Pybricksdev and Pybricks Code.

The firmware should then start the first script rather than looking for the one called `__main__` and always finding it in the first spot.

**Context**
The name is needed to display it in the EV3 UI. We can't have all them called `__main__.py` 😄

dlech, 2025-09-30:
OK with me.

laurensvalk, 2025-10-06:
👍 

> We can work around this by making a file literally named __main__.py if we ever need to do a git bisect and run a program on the hub.

Nice one. Hadn't thought of that.